opentx/opentx #2793: V2.1 telemetry calculated sensors: ‘Totalize’ vs ‘Consumption’

pindeslandes, 2015-08-28:
The way I understand the ‘Totalize’ function is that it integrates the selected source over the time (I am just assuming, as I haven’t upgraded my Taranis to 2.1 yet and I couldn’t find any information about it anywhere).

- If so, I can’t see the difference with ‘Consumption’ because this is what ‘Consumption’ does with the current.
And if ‘Totalize’ and ‘Consumption’ do the same things, couldn’t these two functions be merged?

- But if I am wrong, meaning ‘Consumption’ is the only one that integrates the source, couldn’t this function be extended to other sources (and be renamed as ‘Integrate’ or something like that)?

kilrah, 2015-08-28:
Consumption is integrated over time, while totalize just adds each value as received. Was made for someone who is designing a fuel sensor.

pindeslandes, 2015-08-28:
Thank you for the explanation.
So, what about extending 'Consumption' to other sources than the current?

kilrah, 2015-08-28:
What would it make sense with?

pindeslandes, 2015-08-28:
A few example I have in mind:
- Consumed energy from Voltage and Current sensors.
- Air distance from Air speed to find the best Cz/Cx of a glider.
- Heading from Accelerometer.

kilrah, 2015-08-28:
The first one why not, the other 2 I wouldn't expect to be anything close to usable with the precision and the slow and varying update rate we have. It's already hard to have anything remotely reliable when we've got 100Hz update rate...

pindeslandes, 2015-08-28:
Actually I think the 100 Hz rate would be fine enough for the second one, because when you try to find the best Cz/Cx, you don’t shake your glider.
Anyway, I was assuming that other people would think of other and smarter uses of an Integrate function and I was also supposing that if the function can be opened to one other source (Voltage x Current) it can be opened to any (up to the user to deal with the actual significance of the result).

[That gives me the idea of a 'Derivate' function. I will issue it.]

kilrah, 2015-08-28:
> Actually I think the 100 Hz rate would be fine enough for the second one, because when you try to find the best Cz/Cx, you don’t shake your glider.

That's what I mean, to have 100Hz you need to do the processing onboard. On the radio we onyl get 2-5Hz, and the rate varies. Doesn't make much sense doing any of that on the radio, you'd rather repurpose a flight controller and all its sensors, do the calculations onblard, and send down the processed values as standard telemetry.

pindeslandes, 2015-08-28:
Oh, I had no idea of that 2-5 Hz limitation. Actually I was convinced it was at least 10 Hz because it’s the maximum rate of data logs… Not sure it wouldn’t be usable for a glider I would let go straight for 20-30 seconds, though (my concern is the initial setting of a new glider).

And, unfortunately, *repurposing a flight controller and all is sensors and reprocess* is far beyond my skills, I am afraid!

projectkk2glider, 2015-08-28:
@pindeslandes I use Lua script that filters and averages the vario data to tell me the descend ratio of my glider. Much more powerful.

closed 2015-08-28 by kilrah

kilrah, 2015-08-28:
As Totalize was not restricted to a specific sensor type and neither was companion for both totalize and Consumption I've removed the filter. Looking forward for your tests ;)

pindeslandes, 2015-08-31:
Thank you.
I am not sure to understand exactly what ‘remove the filter’ means but it sounds good enough!
About testing, honestly, before deciding to upgrade to 2.1, I will wait a few months for it to be more stable (but I try to help with issues with Companion).