opentx/opentx #2794: V2.1 telemetry: a few bugs

pindeslandes, 2015-08-28:
Companion or simulator only, not checked on Tx.

- ‘Totalize’: source unavailable (only in Companion, not in simulator)
- °F : precision unavailable
- Companion (french) : ‘Barre de titre de **l’céran** d’accueil’

Not really bugs about units (Companion):
- mAh is not an energy
- ° is not necessarily a heading
- dBm is not necessarily RF related

pindeslandes, 2015-08-28:
I can’t see Precision either for calculated physical values (Distance, Cell and Consumption). Is it because it defaults to the source’s precision?

kilrah, 2015-08-28:
All values are handled internally as integers, and precision is "how many digits to shift the comma". It's processed before calculation, so if you changed it on anything calculated the result would just be wrong Including with °F which is why it is hidden.
*In our context* dBm and ° always relate to RF level and heading.

pindeslandes, 2015-08-28:
Thank you.

By the way, in Companion, just at the right of ‘Temp (°F)’, you can see ‘Precision Ratio’ instead of ‘Ratio’.

About the *context*, I’m sorry I am not sure to understand: if I want to display, say, a pitch, I will use °, won’t I. So it is just that ‘Heading’ should be understood as ‘In particular for heading’, right?

kilrah, 2015-08-28:
Well that wasn't part of the FrSky system :)
OK, actually all the "descriptions" in the units should be removed doesn't make much sense having them.

commit 3920cc69 (2015-08-28): Fix #2794
French translations
closed 2015-08-28 by kilrah