opentx/opentx #2797: Turnigy 9XR-PRO/restore model

asykora, 2015-08-28:
Model does not recover from an SD card.
Firmware: opentx-9xrpro-autosource-battgraph-cz-2.1.2N427.bin

bsongis, 2015-10-13:
Did you try to restore a model that you just backuped? Would you attach the file of the backup?

asykora, 2015-10-13:
> Komu: opentx/opentx <opentx@noreply.github.com>
> Předmět: Re: [opentx] Turnigy 9XR-PRO/restore model (#2797)
>
> "

Yes. The transmitter creates a new model.

---------- Původní zpráva ----------

(https://github.com/opentx/opentx/issues/2797#issuecomment-147668615). 

"=

kilrah, 2015-10-13:
That does not give any detail...

asykora, 2015-10-14:
Video of the issue: https://youtu.be/FcQ_3yH1nKs
(https://youtu.be/FcQ_3yH1nKs)

---------- Původní zpráva ----------

Datum: 13. 10. 2015 16:02:01

(https://github.com/opentx/opentx/issues/2797#issuecomment-147723943). 

"=

kilrah, 2015-10-14:
Video is private, can't be watched.

asykora, 2015-10-14:
The video is public. Please try again.

---------- Původní zpráva ----------

Datum: 14. 10. 2015 10:53:34

(https://github.com/opentx/opentx/issues/2797#issuecomment-147980444). 

"=

kilrah, 2015-10-14:
Still doesn't answer the questions we asked...

Are those files backups you just made, on this radio, and with the same OpenTX version? If not where do they come from?

asykora, 2015-10-14:
Backups are created in OpenTx 2.1.3.

---------- Původní zpráva ----------

Datum: 14. 10. 2015 11:28:48

(https://github.com/opentx/opentx/issues/2797#issuecomment-147988803). 

"=

kilrah, 2015-10-14:
On the same radio? Same type? Please give all details.
A Taranis backup would not work on a 9XR-Pro or any other type, and vice versa.

asykora, 2015-10-14:
Transmitter Turnigy 9XR-pro.
Backups created on 9XR-pro.

The problem started from OpenTx 2.1.0

---------- Původní zpráva ----------

Datum: 14. 10. 2015 13:04:23

(https://github.com/opentx/opentx/issues/2797#issuecomment-148017542). 

"=

projectkk2glider, 2015-10-14:
Backup/restore works on current `master` without any problems for me in `simu`. I find it hard to believe it would work differently on the radio.

griebd, 2015-10-14:
@asykora, your issue sounds like a user error... to help you we need detailed data!

To begin with, from where did you get those backups? You made them? Or copied from somewhere?
If you did them, on which radio and which firmware?
From your posts it appears that you made those backups on a 9XR-pro running OpenTx 2.1.3, is that right? I actually don't believe it is because you later say that "The problem started from OpenTx 2.1.0", and on your first post you say you are using firmware version 2.1.2N427. Why would you install 2.1.3 make a backup and go back to 2.1.2N427, a nightly test version????

It would be a lot helpful if you could upload those backups somewhere so we can have a look on them!

You may have found some rare strange bug, but we need data to work it out!!!! We try to do the same and it works, no one else is complaining...

asykora, 2015-10-15:
On the SD card, I deleted all backups.
On the transmitter I deleted all models.
New video: https://youtu.be/SF6CJGhLZKM(https://youtu.be/SF6CJGhLZKM)
Firmware:  OpenTX 2.1.3

---------- Původní zpráva ----------

Datum: 14. 10. 2015 14:29:58

(https://github.com/opentx/opentx/issues/2797#issuecomment-148033422). 

"=

asykora, 2015-10-15:
New video

https://youtu.be/SF6CJGhLZKM(https://youtu.be/SF6CJGhLZKM)

---------- Původní zpráva ----------

(https://github.com/opentx/opentx/issues/2797#issuecomment-148033422). 

"=

asykora, 2015-10-15:
New video
https://youtu.be/SF6CJGhLZKM

projectkk2glider, 2015-10-15:
From the last video: the model name is not restored. Perhaps that is intended. Have to check...

projectkk2glider, 2015-10-15:
No, on the `simu` the name is also restored. I checked booth 2.1.3 and current `master`. Why would this work differently on the radio?

projectkk2glider, 2015-10-15:
@asykora please upload ABC.bin backup from you SD card. You will find it in MODELS directory.

asykora, 2015-10-15:
ABC.bin in Annex

---------- Původní zpráva ----------

Datum: 15. 10. 2015 13:33:25

(https://github.com/opentx/opentx/issues/2797#issuecomment-148360686). 

"=

projectkk2glider, 2015-10-15:
@asykora  you can not upload attachments to gitgub like that. Upload your file to some file sharing service (like http://wikisend.com/) and post link here.

projectkk2glider, 2015-10-15:
@asykora Additional question: when you restore the model do other settings (except the model name) restore or not. For example do the mixes (other than default ones) restore or not?

asykora, 2015-10-15:
Does not restore other settings. Creates a basic model.

---------- Původní zpráva ----------

Datum: 15. 10. 2015 15:02:34

(https://github.com/opentx/opentx/issues/2797#issuecomment-148377799). 

"=

projectkk2glider, 2015-10-17:
@asykora We are still waiting for your BIN file. Upload your file to some file sharing service (like http://wikisend.com/) and post link here.

asykora, 2015-10-17:
http://www.uschovna.cz/zasilka/GLSSAZFZ58BHD6GM-JSP/ 

---------- Původní zpráva ----------

Datum: 17. 10. 2015 20:01:08

(https://github.com/opentx/opentx/issues/2797#issuecomment-148938259). 

"=

projectkk2glider, 2015-10-17:
OK your file restores successfully on `simu` (at current master state). I am out of ideas. @opentx/commiters  Anybody have this radio and is able to reproduce this?

griebd, 2015-10-17:
@asykora can you post a screenshot/photo of your radio on the firmware version page? Long press '+' and then four clicks on '-'!
Have you compiled the firmware or downloaded it through companion?
If self compiled, with options did you use?
If downloaded in companion, with options are selected? Menu Settings -> Settings... -> Build Options in Radio Profile!
Thanks.

bsongis, 2015-10-17:
I have the radio, I will try to find one minute tomorrow.

asykora, 2015-10-17:
Used OpenTx Companion 2.1.3

Foto :

http://www.uschovna.cz/zasilka/GLHYJ77KHTI8NX8B-HTD/

---------- Původní zpráva ----------

Datum: 17. 10. 2015 21:55:55

(https://github.com/opentx/opentx/issues/2797#issuecomment-148947650). 

"=

griebd, 2015-10-17:
From the photo/screenshot, just to make it easier:

radio:
FW: opentx-9xrpro
VERS: 2.1.3
DATE: 2015-09-17
TIME: 12:36:18
EEPR: 217

companion options:
Radio: 9XR Pro
menu: cz (shouldn't the words data and time been translated on the radio? they aren't)
selected options: autosource, autoswitch and battgraph

bsongis, 2015-10-18:
Reproduced here.

commit 9463f49b (2015-10-19): Fixes #2797 - Model Restore broken on Sky9X / 9XR-PRO

It seems I am not allowed to write the 4bytes header separately. It's
overwritten by 4 zeroes at the next write.
closed 2015-10-19 by bsongis