opentx/opentx #2798: Updated nl.h.txt

fraternl, 2015-08-28:
In special functios the "reset ..." was using text it got from totally somewhere else.
My translation of "Telemetry" into "Telemetrie" messed up the string that was used as an array.

http://hmvc.eu/nl.h.txt

Since this recently discovered issue with jitter was solved I suspect a release of 2.1.2
I hope to get this in on time.

projectkk2glider, 2015-08-28:
I am on it.

projectkk2glider, 2015-08-28:
This does not look right to me. The length is 10 characters:
```
#define LEN_VFSWRESET          TR("\004", "\012")
```

But you now you translated it to 9 chars long:
```
  #define TR_FSW_RESET_TELEM   TR("Telm", "Telemetry")
```

The error must be somewhere else.

fraternl, 2015-08-28:
Sorry... missed that....

But the part where it goes wrong is exactly the same as the english counterpart...
I don't think it's something wrong with the firmware itself.
Must still be something in my nl.h.txt   
I will look at it this evening..  have to work now

projectkk2glider, 2015-08-28:
I've found it, other strings were too short. Fix coming.

closed 2015-08-28 by projectkk2glider

fraternl, 2015-08-28:
Ah... thanks....
Yours is better...

commit 5a049857 (2015-08-28): Fixes #2798: Dutch translations fixed