opentx/opentx #2799: Wish:  Importing models on radio for sharing with others on the field.

fraternl, 2015-08-28:
Currently when I'm saving my models in Companion I'm saving them in the SD-structure on my harddisk in the folder /eepe
Because it's synchronized using Companion these eepe's also reside on the SD-card of the Taranis.

I just made a back-up of the EEPROM and tried to select one of those eepe's and see what it would do.
But no option was given to read that eepe
I could of course do it with a back-up, but I wasn't looking for a complete replacement of my model settings. 

I was aiming for the addition of the models that are in that eepe without overwriting the radio settings.
A downloaded eepe from rcsettings could also be on that SD-card.
Maybe even a model-picker afterwards and then let that model go after the last model of the current setup (even preferable than the import of all models).

If this is all too hard to implement this on the radio you can just close it

kilrah, 2015-08-28:
You can backup your entire eeprom and reload it, including downloaded ones right on the radio as long as they're in .bin format instead of .eepe.

You can then "backup" and "restore" individual models too, so I don't really see what you're missing. On the field anyone can "backup" their model, give you their card, you restore, done.

fraternl, 2015-08-28:
I can't load someone else's eeprom without losing my own settings, can I?
I always assumed I can only restore an eeprom as a whole (all models and radiosettings are overwritten).

So I can go to the field and give my SD-card to someone and he can put it in his radio and can get my models without losing his own?

kilrah, 2015-08-28:
Of course if you load the entire eeprom you lose what you had, but it's simply not the correct thing to use...

Again you would "backup" the models you want to give him on the model selection page, and he would "restore" those on empty slots.

fraternl, 2015-08-28:
Hi, André
I just tried what you said and that works.....
It's already there...

I can save (backup) a model to the SD-Card and take it out... The other can do a "restore model" if he long presses on an empty slot.

Thanks....

closed 2015-08-28 by fraternl