opentx/opentx #2800: Von 2.0.17 nach 2.1.1

frobert0, 2015-08-28:
Nach Update und EEProm Konvertierung ist der Timer1 Schalter von !SG^ in !FM8 geändert worden.

In Mischer und anderen Teilen wo !SG^ verwendet worden ist hat es funktioniert.

Wieso ist bei der X9D mit 2.1.1 Timer3 im Sender nicht vorhanden und in Companion 2.1.1 schon

kilrah, 2015-08-28:
Conversion done on the radio or in companion?

To have timer 3 on the radio you need the "timer3" firmware option.

frobert0, 2015-08-28:
Conversion in Radio von 2.0.15 to 2.1.1

Firmware 2.1.1 in Radio

???? !SG^ to !FM8

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

kilrah, 2015-08-28:
Hmm for me on a conversion from 2.0.17 to 2.1.1 I get !SG^ -> !SFv. No !FM8.

commit f0df8450 (2015-09-06): Fixes #2800
commit a18ff7b5 (2015-09-06): Fixes #2800
closed 2015-09-06 by bsongis