opentx/opentx #2803: lua script for V2.1

MRTJ30, 2015-08-28:
Hi Guys I realized that the V2.0 lua script is not compatible with V2.1.1 firmware. Is there something you guys could do to make it so the new firmware would recognise old version of lua or could you write new telementry script for things like Vector with a compas indicator of the model on the right half with model battery and rssi and tx signal on left. Thank you

projectkk2glider, 2015-08-30:
>Is there something you guys could do to make it so the new firmware would recognise old version of lua

No way, the new Lua interface is just that, new, different. It is a consequence of the totally changed telemetry system in 2.1. Nothing can be done here. The scripts will need adaptation to 2.1. The script itself can get the OpenTX version if needed.

>ould you write new telementry script for things like Vector with a compas indicator of the model on the right half with model battery and rssi and tx signal on left. 

We pretty much don't provide any Lua scripts. That is usually the work of other users. Ask the original script author to do it.

closed 2015-08-30 by projectkk2glider

MRTJ30, 2015-09-13:
Thank you guys I don't think I need the Lua script any more only just realized how power full the new firmware is and only just realized that there is hidden menus and the telemetry menus and setup is different but great