opentx/opentx #2804: Telemetry logging

flarssen, 2015-08-29:
I had an issue with a hexacopter flipping violently to the right or left. I suspected that it might be caused by a  tired Lipo and ESC cutoff enabled by accident. I had logging enabled, so I hoped to find the cause based on the log. I checked the ESC settings (Plush 6A, BlHeli FW), and cutoff was disabled for all six. Log showed that VFAS dropped low enough to influence BEC voltage. The actual issue is that the logging of the controls seems to be affected by this low voltage situation on the RX side. The copter was hovering with little stick input. Still I see  violent changes in stick data. How can that be?
Running 2.1.2N431.
Log file; https://www.dropbox.com/s/1gbvn6czhjrxyyx/low.log?dl=0

[image]

kilrah, 2015-08-29:
> The copter was hovering with little stick input. 

That's exactly what you see here, with stick movements of about 1/10th of their throw on either side.

closed 2015-08-29 by kilrah

griebd, 2015-08-29:
Just to make it clear, the plotting is adjusted so you have "maximum" resolution for you data. In your case, as there was little stick input, you can see this input really detailed. As @kilrah sad, that is noted on the left axis scale in your plot.