opentx/opentx #2806: Telemetry screen: Firmware 2.1

rafabene, 2015-08-29:
In companion I can set more than one Telemetry screen, but when I'm execution the firmware it shows only one screen.

kilrah, 2015-08-29:
Can't reproduce. What radio type? What screen type? What exact version? Please detail your issues.

rafabene, 2015-08-29:
It's a 9xrPRO with 2.1.2N

bsongis, 2015-09-02:
First there is a problem in menu navigation when there are hidden items!

bsongis, 2015-09-02:
Would you be able to test my fix tomorrow on the nightly build and tell me if it is better?

rafabene, 2015-09-02:
@bsongis Who was this question for ?

bsongis, 2015-09-02:
For you :)

rafabene, 2015-09-03:
I can test it. I'll will do it today at night.

rafabene, 2015-09-03:
I should try today again together with #2824​

rafabene, 2015-09-04:
I tested N440 and 9xrPRO still showing just one telemetry screen

bsongis, 2015-09-05:
I just tested, no problem on my 9XR-PRO
Do you see your screens in the Telemetry screen (Model menu)?
Do you press [Down] to switch between all those telemetry screens?

rafabene, 2015-09-06:
In my Telemetry screen (Model menu) I have two screens configured but when I press [Down] to switch between those telemetry screens, I can see just the first one. I tried it with different models

bsongis, 2015-09-06:
I tested with numbers, perhaps you have bars on one screen? Is it the same if you create your screens on the radio. Just to know if it is a Companion or a firmware bug ...

bsongis, 2015-09-06:
Would you send me your eeprom (export from Companion). I can't reproduce.

kilrah, 2015-09-06:
Can't reproduce either.

rafabene, 2015-09-08:
You can get it here: https://www.dropbox.com/s/6b8xl9jon8f5pkg/Modelos%202.1.bin?dl=0

kilrah, 2015-09-08:
Your 2nd screen has bars configured with a source but with no range (From 0 to 0). That is equivalent to nothing, so it is skipped.

closed 2015-09-08 by kilrah