opentx/opentx #2809: Companion conversion issue

janekx, 2015-08-30:
somwhere I found some servo tester epee so I try to test it. 
At Companion 2.0.17 and working

[image]

[image]

[image]

But opening at 2.1.1N434 

[image]

[image]

[image]

As you can see the channels change from Ch24 to CH23 ate mixers, from MAX to S1, other chanenels change numbers too and at logical switches GV1 change to CH32. 
Can be the issue from how does loading progress and filling the values some channels or GV are not usable? interesting MAX there should not be any reason to change.

kilrah, 2015-08-30:
To do together with #2788

bsongis, 2015-09-01:
Would it be possible to have your eepe file?

bsongis, 2015-09-01:
This is not related to #2788 as I am almost sure it's a conversion issue from OpenTX eeprom < 2.0

janekx, 2015-09-01:
Will upload it as soon as will be close to pc

bsongis, 2015-09-01:
Would you send it by email?

janekx, 2015-09-01:
Yes will send  it by email to you as will be close to PC if not possible to commit it here.

closed 2015-09-02 by bsongis

janekx, 2015-09-05:
Confirming that now it translate well. Thanks

commit f7c9da68 (2015-09-02): Fixes #2809