opentx/opentx #2810: Ghost sensor:   On one of my models I suddenly have a telemetry entry 1110

fraternl, 2015-08-30:
In the past I had duplicate Vario's in one of my models.
I suspect it happened because I used another one of my 4 vario's

To prevent this from happening again I enabled the option "ignore ID's"  (is that a correct solution, BTW?)

Now I all of a sudden have this extra telemetry sensor.
Most of the time I edit only 1 model and copy that model to the other 3 after I do a restructuring of the setup.
Afterwards I will edit the Name, outputs, RxNum, and GVARs  (I'm telling you all this to give you additional information).

I regularly backup as an eepe and it turns out this sensor appeared after I flew that model last Friday. I started flying with

Addition:

I added that same model (of Thursday's setup) without the extra sensor to my setup using Companion and made a backup of the EEPROM.
I then connected it to the model.
This extra sensor didn't appear.
No telemetry sensors were swapped this weekend.

[image]
[image]
[image]

fraternl, 2015-08-30:
I just noticed another ghost sensor in the 2nd "Minimoa"

projectkk2glider, 2015-08-30:
Does this ghost sensor give any data, does it change? Which sensors are connected?

fraternl, 2015-08-30:
On the DLG only the variometer is connected (jk2), on the motorglider I have the 40A current sensor together with a variometer.

No, it's not giving data.

projectkk2glider, 2015-08-30:
>To prevent this from happening again I enabled the option "ignore ID's" (is that a correct solution, BTW?)

No, that option is for bad sensors that answer on every device address and therefore fill up all available 32 sensor slots. It won't help in your case.

I would say that your non FrSky vario sensor is doing this and that OpenTX is not to blame here. I can't find any info on jk2. Is it open source?

projectkk2glider, 2015-08-30:
And the id of 0x1110 that you are getting is very close to the official vario id 0x0110.

fraternl, 2015-08-30:
JK2 is the name of the model that's having this ghost sensor.
The Minimoa is the other.

projectkk2glider, 2015-08-30:
Uff, I have to drag every bit of information from you :smile: 

Is the vario from FrSky (which one exactly) or from third party? Which receiver is used? Which data value is shown for sensor 0x1110?

fraternl, 2015-08-30:
They are all Frsky....
All models have the Vario
The Minimoa also has the 40A current sensor

X4R for the 4 DLG's (Cumel, JS, JK, JK2)
The X6R for the Minimoa

No data value... (haven't you loaded any of the posted EEPROMs????)

fraternl, 2015-08-30:
Maybe leave things for now and treat it as a fluke...?
I will delete the sensors and maybe someone gets this too

kilrah, 2015-08-30:
I would think so. First time I hear of such a thing.

lepicfoto, 2015-08-31:
actually I also have "ghost " in my setup, in my case, the " 2 X VFAs + 2X Curr " and I have only one VFAs in my setup, gotta go jobbing, more latere, just to let you know you are NOT alone with your ghots's :-)

Sacre100, 2015-08-31:
Another case (not mine) beginning August : http://frskytaranis.forumactif.org/t2459p105-presentation-de-la-telemetrie-opentx-2-1-traduction-de-opentx-2-1-telemetry-system-preview#28558

projectkk2glider, 2015-08-31:
All examples of ghost ids have only one bit different from the correct one. Is it possible that checksum is letting these erroneous packages trough?

fraternl, 2015-08-31:
> I would think so. First time I hear of such a thing.
The first reaction would be to just delete it and go on....
I decided to post it and maybe trigger an "aha, I have this too"

Before a sensor is added some more validity checks could be done....
But getting to the source of the problem is best. 
You guys are good at that I noticed in the past.

flarssen, 2015-09-01:
Just for the record and maybe related; I get a filled up sensor list with a D series Rx too. Possibly caused by feeding data to the Rx too fast, leading to data corruption. That could be blamed on the Rx firmware, no reason it should send out data for ID's not defined within the HUB protocol specs.
I also had duplicate SWR and RSSI entries at different ID's (if I remember correctly), but didn't investigate it further.

kilrah, 2015-09-01:
> That could be blamed on the Rx firmware, no reason it should send out data for ID's not defined within the HUB protocol specs.

No reason it would limit downlink to the hub format. The receiver doesn't filter anything, FrSky advertised the possibility to send down whatever you wanted using 3rd party equipment.

> I also had duplicate SWR and RSSI entries at different ID's

Different receiver models use different IDs, so if you played with different RXs on a same model you'll have duplicates.

fraternl, 2015-09-01:
Different receiver models use different IDs, so if you played with different RXs on a same model you'll have duplicates.

I would assume he mentioned that (different RX's on same model) if he did.  Did you, Fred?

fraternl, 2015-09-01:
If the system would "play hard to get" and need at least 3 offerings of data before getting added to the inner circle of trust?
I don't know the S-port protocol. If it's just sending data packets with each one containing an ID it is likely that if the ID is corrupt, the rest of the data is somehow corrupt too.
A rejection of those packets could be enough...

Just speculating here and maybe giving ideas...

flarssen, 2015-09-01:
I have only used one Rx for this model, but I will see if it can be reproduced. 
I don't think three offerings would work since I often see same ID reappear after deletion. 
I should know better than suggesting limits on HUB data, since I'm having great benefit of the unspecified 0x39!

lepicfoto, 2015-09-01:
Hi everybody,
I've looked a little at my setup, and I use only genuine FrSky sencorer.
I just redid my J3, connected everything up, turned on everything, and then bound a FrSky X8R Receiver to Taranis X9D.
There are still two sencorer which is dublicat, it is the FAS-100, I think,
it is connected via the vario HP data in.
I have put a few files, A "std telemetry.eepe" and a list of the sencorer as they are found on X9D,"x9d telemetry.ods", on drop box, they are all raw from the ground up,
so there is no noise from other processes,
They are located at:

https://www.dropbox.com/sh/oyn8p7osvxl2r47/AADA50F7WnuoOZjDLjqcICYRa?dl=0

just to provide some data. Then we can analyzed the problem, when the time is for it :-)

PS, the proposed method, just to delete the sencoere that are Ghost sencorer
works perfectly fine. If one has suffered knowledge in what goes on in the X9D.

:-)
Jesper Malling
Denmark.

flarssen, 2015-09-02:
Regarding my duplicate RSSI/SWR:
https://github.com/opentx/opentx/issues/2762

kilrah, 2015-09-02:
?

Completely unrelated.

projectkk2glider, 2015-09-03:
@lepicfoto has this ghost sensors:
* VFAS: 4 x id 0x003B with different instances: 0, 1, 1(duplicated) and 2. I can't see how the second instance 1 was created. Did you created it manually?
* Curr: 3x id 0x0028 with instances: 0, 1, 2 and 1 x id 0x0029 with instance 2

Basically all above have only one bit changed either in id or instance, only one has two changed bits: that is the Curr with id 0x0029 with instance 2.

I still believe that the checksum is letting this errors go trough undetected.

projectkk2glider, 2015-09-03:
I see no other reason but a line noise that produces these bit errors which then sometimes pass trough existing checksum. The checksum algorithm is very simple and will eventually (with enough corrupt packets) pass trough an invalid data packet.

We can not change/improve the checksum, it is a part of FrSky Sport standard and encoded in all their hardware (sensors, receivers, tx modules..).

So my proposals are:
* we add a proper validation for DATA_ID byte which uses bits 7-5 for validation.
* we add one temporary holding sensor slot which will be used for adding new sensors. 

The new temporary holding sensor slot will work like this:
* we receive a packet from not yet known sensor
* if the holding sensor is empty we initialize it with new data (sensor id and instance, protocol) and set its expire time.
* now we wait for more packets from that sensor.
* every time a new packet from that sensor arrives we reset expire time and increment good packet count (in the temporary holding sensor slot)
* if the count of good packet reaches predefined threshold value, we consider this sensor to be real and we add it to the model as usual. The temporary holding sensor slot becomes free.
* if the temporary holding sensor slot expires, then we just fee it.

This way recognition of new sensors will be slower, because each one will have to be validated before and that could take a couple of seconds. It will depend on the parameters we set:
* the number of good packets required 
* the expiration time (the longest time between two packet from the same sensor + some reserve)

projectkk2glider, 2015-09-03:
We could also keep count of bad (those that fail checksum) and good packets and display that number to the user somewhere. It might help if this is a problem of a bad wiring and/or interference.

lepicfoto, 2015-09-03:
Hello to all.
thanks for the effort.

Reply to projectkk2glider.

Everything is connected to a X8R receiver, and then the receiver is bound to a Taranis X9D, NOT A Taranis X9D +.
it is connected in series to follow their address as read from the "FrSky SBUS Servo Channel Changer" The sequence is:
ID-1, Vario high sens + FAS-sensor 100 Amp
ID-2 Lipo volt sens
ID-4 GPS Sensor
ID-5 RPM Sensor.
Everything is connected on my "desktop" and is supplied with a 6V 5Ah NiMh.
After X8R has been bound to X9D, the sensors show up as pr the list, and that is where I have reached, I have not created anything at all, (I need to learn how to do it manually) so I am a bit confused but I'm learning all the time :-)
thanks for the effort. you really rock, doing a giant great work,
Greatest respect from,

Jesper Hansen malling
Denmark

bsongis, 2015-09-03:
I do like the "new temporary holding sensor slot" idea. Just a couple of comments:
- if the holding sensor is empty** or different** from  we initialize it with new data (sensor id and instance, protocol) and set its expire time.
if the count of good packet reaches predefined threshold value, we consider this sensor to be real and we add it to the model as usual. ** The temporary holding sensor slot becomes free. <= not needed! **
**if the temporary holding sensor slot expires, then we just fee it. <= not needed!**

projectkk2glider, 2015-09-03:
No Bertrand, imagine a case of a new model with two sensors A and B. We get alternating packets for each sensor. Overwriting *temporary holding sensor slot* every time we receive a sensor packet would not work in such case:
* receive packet A -> A into temp slot
* receive packet B -> B into temp slot
* etc...

The process would never finish. My case would work:
* recieve packet A -> temp slot empty -> A into temp slot
* recv. packet B -> temp slot busy -> discard
* recv A -> inc temp slot count
* recv B -> discard
* etc...
* recv A -> count big enough -> Create sensor A, free temp slot
* recv B -> B int temp slot
* recv B -> inc tmp slot
* etc...
* recv B -> count big enough -> Create sensor B, free temp slot

projectkk2glider, 2015-09-03:
Looking at the eepe from @lepicfoto again I see different data (what was I seeing before??):
* VFAS and Curr are detected 4 times:
 * 3 times just the instance changes from 0, 1 and 2.
 * Then a very peculiar thing: we have a **duplicate** VFAS and Curr with instance 2!!! I can not explain how this happened. According to @lepicfoto he has not manually entered anything.
  * the id's of those sensors are always correct

So his case might be something else altogether, could his equipment actually generate VFAS and Curr with multiple instances?

bsongis, 2015-09-03:
Ok @projectkk2glider  you are right. I didn't think of this use case with all sensors discovered at the same time, which is the case when you start your model.

bsongis, 2015-09-04:
@projectkk2glider do you take it? otherwise I will :)

projectkk2glider, 2015-09-04:
You take it, I am busy today.

bsongis, 2015-09-04:
Not today for me as well, but perhaps this week-end if weather is bad :)

Another idea could be to add a "Reject new sensors" checkbox. It would even allow the user to remove some sensors he doesn't want to see (otherwise they respawn each time the sensor sends something)

projectkk2glider, 2015-09-04:
Good idea.

fraternl, 2015-09-04:
Would such a "reject new sensors" checkbox stop all other efforts to detect new sensors?
If that's the case why not the reverse?
Have a period to detect new sensors and that way make that a procedure.
Otherwise it just leads to confusion.

fraternl, 2015-09-05:
No reaction?
It's not a bad idea to let the discovery process only take place in a small window immediately after it first received a valid link (or maybe not even then) but for sure when the radio is on the telemetry page.
Any strange sensors can get immediately deleted if they pop up.

As far as I can gather there is no use for sensors that are discovered using flight.

Yes, it would need to be known by the users, but if there's any time to introduce such a "special procedure" it is now, when 2.1.x is still in its experimental phase.

kilrah, 2015-09-05:
I'd rather want to find the source of the problem, as in theory it simply should never happen...

fraternl, 2015-09-05:
@kilrah ...I totally agree
But if it's not found I prefer my alternative above a checkbox that has to be set on/off each time.
What I'm proposing is in fact more or less the same.
The only difference is that there's no checkbox, but the option is set on/off by selecting the telemetry page.

lepicfoto, 2015-09-05:
I will have to support Kilrah,
I as a user do not like that there's something not right with my radio.
I am well aware that it is a mega job.
Future users may should not get the opportunity to call our radio system full of errors, it is a situation that occurs really fast, and without looking at the system as a whole,
Many large companies has ceased to exist over the minor errors that they have made "work a round" for.
I'm pretty proud to be one of opentx users, especially when I'm in the club, and others see the possibilities inherent in our system .

with great respect
Jesper M

kilrah, 2015-09-05:
Has someone actually been able to relatively consistently repeat it with a given setup? Any of those of you who had 2-3 occurrences?
I've been using the 2.1 telemetry for several months now including when it was in development and never had any unexplainable sensor come up like this...

bsongis, 2015-09-05:
The same here. But it's really possible that the checksum is not able to filter all transmission errors, thus such a mechanism. I will implement @fraternl idea though. It will allow a user to hide sensors he doesn't want to see (as those I have on my naze32 with baseflight).

kilrah, 2015-09-05:
> But it's really possible that the checksum is not able to filter all transmission errors

Indeed, and to be fair I haven't used a lot of telemetry in actual flight so far, mostly bench use. The probability of transmission errors is obviously higher in actual use.

lepicfoto, 2015-09-05:
Hi all
I just made some NEW "models" for you
they are in my dropbox at: Link
https://www.dropbox.com/sh/oyn8p7osvxl2r47/AADA50F7WnuoOZjDLjqcICYRa?dl=0

there is a txt file which should help with the decoding of my eepe files names.
I started with a completely empty model, mod2-ntl, and  straightforward I have added sensors one by one, by their address.
I've only had time to set it up on my desk, and it's not been flown yet, and no tel screens set up as of now,
Jesper M

Sacre100, 2015-09-05:
I don't know if it's related but on my Taranis (not plus) OpenTx 2.0.17 french, I get many time from telemetry unrealistic data. I have always suspected the sensor (Frsky GPS or Frsky Vario high precision) but perhaps the problem is on TX side. Here is an example with a GPS Speed of 44'634.362 kts (not bad for a 4m glider) GPS and vario were connected : 

[image]

bsongis, 2015-09-05:
That's very possible. In OpenTX 2.0.X such a problem would add a very wrong value into the others values (the whole range of 16possible IDs for each sensor was merged into only one value).

projectkk2glider, 2015-09-06:
I like the last solution (that was @fraternl idea) to have an option to enable/disable discovery of new sensor. It will elegantly solve several problems.

I would still like to see a counter for good/bad telemetry packets.

fraternl, 2015-09-06:
Although I was credited by Bertrand, the manual disabling of discovering new sensors was his own.
I then proposed the idea to make that option automatic by turning on the discovery after receiving a link  for a while (2 minutes?) and while it's on the telemetry page.

Could the logging of unprovisioned sensors get into the CSV or is that difficult to implement?
A seperate column for unprovisioned and just drop values when they come...

bsongis, 2015-09-06:
I will prefer a counter as @projectkk2glider says. Perhaps even different counters (wrong CRC / trashed frames / ?)

bsongis, 2015-09-06:
@opentx/commiters We will need to release tomorrow, FrSky needs it for the next X9E batch. Would you take care of the translations, this feature adds some new strings. Thanks!

closed 2015-09-06 by bsongis

fraternl, 2015-09-06:
@Bertrand   If the discovery of new sensors is enabled when you're in the telemetry page on the radio it  would be "best of both worlds".
Although I would agree it then isn't doing what it says it does.

kilrah, 2015-09-06:
No it now requires manual intervention to enable discovery. That way you can prevent unwanted sensors to come up.

commit 4b9fcdaf (2015-09-06): Merge pull request #2832 from opentx/bsongis/#2810_reject_new_sensors

Fixes #2810 - Reject new sensor option
Fixes #2730 - Delete all sensors
commit b54123f8 (2015-09-06): Fixes #2810 - Reject new sensor option