opentx/opentx #2812: Output limit test mode

janekx, 2015-08-30:
I got idea how to simplify check of the limits and center.
Make on outpouts screen some test mode where when is activated will be supressed all mixes, curves and so and any one can easily checks limits. 
It can be helpful when you forget to check before adding all mixes and not sure your settings will not harm your servos in some conditions or when you must replace some servos for other ones and need to checks the limits.

Because it is quite dengerous function it needs to work only on output screen and be deactivated exiting the output screen, but needs full output screen visible. 
Some upper bar info and beeping should be good warning.

kilrah, 2015-08-30:
I don't think we could do anything simpler/quicker than just adding a replace mix line with your favorite control at the end of any channel you want to check.

janekx, 2015-08-30:
Still working on the idea,  will update soon.

kilrah, 2015-08-30:
By the way someone has made a lua script that automatically adds a test line to all channels when run, and removes them when run again.

janekx, 2015-09-02:
Limit test mode. My idea 
On the output page have posibility to swich to test mode. 
Selecting the output channel, holding enter there will be "output limit test mode" as last option.
The window open and there will be possible to select which input will be sended to the selected output. Possible will be gimbals, pots, switches. 
This will assign the input directly to output and user can easily check the limits and PPM center and tune it. If is possible to tune the output in the opened window will be nice to see % and us on all limits and center

Like

Select input ---         -100            0               100
                                1000us     1500us        2000us
Actual                                          0%           1499us       

The selected new limits are long enter saved to the channel. And limit test mode is canceled. 
Pressing exit not save changes and exit. 

Andre 
Lua script is nice but I am afraid that someone can forget that have there test line in mixes.

kilrah, 2015-09-06:
I quite like that.

closed 2020-07-07 by stale[bot]