opentx/opentx #2815: Companion 2.1.1 Fails on mac

rafabene, 2015-08-31:
This is the error report:

```
Process:               companion [10051]
Path:                  /Applications/companion.app/Contents/MacOS/companion
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           companion [10051]
User ID:               501

Date/Time:             2015-08-30 21:46:54.782 -0400
OS Version:            Mac OS X 10.10.5 (14F27)
Report Version:        11
Anonymous UUID:        2D6BF199-B84F-A17B-359E-5CE6F6854AB2

Sleep/Wake UUID:       099929F9-3CA5-4D3A-B935-86B1D1151DF5

Time Awake Since Boot: 71000 seconds
Time Since Wake:       1700 seconds

Crashed Thread:        1  Dispatch queue: com.apple.libdispatch-manager

Exception Type:        EXC_CRASH (SIGSEGV)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Thread 0:: Dispatch queue: com.apple.main-thread
0   QtGui                         	0x0000000100957aa6 QAction::setChecked(bool) + 38
1                                 	0x00000001000d5aea MainWindow::updateLanguageActions() + 762
2                                 	0x00000001000cceaa MainWindow::updateMenus() + 554
3                                 	0x00000001000c6499 MainWindow::MainWindow() + 649
4                                 	0x00000001000cd955 MainWindow::MainWindow() + 21
5                                 	0x00000001000d8849 main + 3577
6                                 	0x0000000100003584 start + 52

Thread 1 Crashed:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8f997232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff8aad5a6a _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8f99694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff900eb3dd start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8f99694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff900eb3dd start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff8f99694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff900eb3dd start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff8f99694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff900eb3dd start_wqthread + 13

Thread 1 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000004  rbx: 0x0000000000000000  rcx: 0x0000000101e7d5b8  rdx: 0x0000000000000000
  rdi: 0x0000000000000003  rsi: 0x0000000000000000  rbp: 0x0000000101e7d640  rsp: 0x0000000101e7d5b8
   r8: 0x0000000000000001   r9: 0x0000000000000000  r10: 0x0000000101e7d5d0  r11: 0x0000000000000246
  r12: 0x00007fff8aaf5258  r13: 0x00007fff73d24ec0  r14: 0x0000000101e7d5d0  r15: 0x0000000101e7d600
  rip: 0x00007fff8f997232  rfl: 0x0000000000000247  cr2: 0x0000000101eff000
  
Logical CPU:     0
Error Code:      0x02000171
Trap Number:     133

Binary Images:
       0x100000000 -        0x1007c8ff7 + (??? - ???) <63BD5910-E079-3937-8AA7-6FD6370D5E13> /Applications/companion.app/Contents/MacOS/companion
       0x1008d1000 -        0x10114eff7 +QtGui (0) <6AB64529-49D5-3ADD-88C0-08139B13E00F> /Applications/companion.app/Contents/Frameworks/QtGui.framework/Versions/4/QtGui
[... 256 lines cut ...]
__UNICODE                           552K
mapped file                       117.6M
shared memory                         4K
===========                      =======
TOTAL                             466.7M

Model: MacBookPro9,2, BootROM MBP91.00D3.B0B, 2 processors, Intel Core i5, 2.5 GHz, 16 GB, SMC 2.2f44
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x859B, 0x43543130323436344246313630422E433136
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x859B, 0x43543130323436344246313630422E433136
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xF5), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.6f3 16238, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 840 Series, 120.03 GB
Serial ATA Device: Samsung SSD 840 EVO 500GB, 500.11 GB
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: IR Receiver
USB Device: Apple Internal Keyboard / Trackpad
Thunderbolt Bus: MacBook Pro, Apple Inc., 25.1
```

kilrah, 2015-08-31:
What were you doing when it crashed?

rafabene, 2015-08-31:
Just tried to open it.

commit 4acebc31 (2015-09-03): Fixes #2815: some languages were not completelly disabled and caused Companion crash
closed 2015-09-03 by projectkk2glider