opentx/opentx #2816: Switch positions in Global Function not detected properly

MarenB, 2015-08-31:
Hey,

I added a 3-pos switch on L that I am going to use for SD-Card logging (Off/1s/0.2s).

While the logging itself works fine, further lines triggering audio messages ("recording on" / "recording off") do not work correctly. Either switch positions are not detected (mid or down) or always detected (!up, even when the switch is in the upper position). 
There seems to be a relation to the special function triggered, if I do not use play audio, but semthing else, it works correctly or not at all, iirc.

I am using OpenTX 2.1.1 on a X9E.

kilrah, 2015-08-31:
Likely related to https://github.com/opentx/opentx/issues/2754

Can you try with a switch less "far" i.e. SI?

MarenB, 2015-08-31:
That sounds very much the same, indeed. Did overlook that existing issue when I searched for it.

Will try it tonight with a "lower" switch. Anyhow, the fact that mainly the NOT-position does not work fits well into the picture of the 64bit index, doesn't it?

kilrah, 2015-08-31:
Hmm actually I don't think the mask is the problem, but problem there is... will continue in the other issue.

closed 2015-08-31 by kilrah