opentx/opentx #2818: V2.1 telemetry: user-definable units (names and sounds)

pindeslandes, 2015-08-31:
These units might be needed:
-hPa and other pressure units,
-Wh and other energy units,
-°/s…
But other people will probably discover other needs and it might become a never ending issue.

So what about the idea of having, in addition to the default pick list and the related system sounds:
-user-definable unit names,
-with those unit names being linkable to a limited number of reserved system sounds (say unit1.wav to unit8.wav).

This way, there would have no more need to take any new imaginable unit into account in further versions of OpenTx.

To create a new unit, the user would have to:
-name it (special characters like µ or ° welcome!),
-link it to one of those unitx.wav system sounds,
-record some relevant sound as unitx.wav.

kilrah, 2015-08-31:
That's going way too far to me. 
So someone will buy some sensor, and will have to configure the unit for it before being able to use it?

I'm tempted to close this, #2817 and revert my change done in #2793, and declare that anything that advanced just has to be done in a lua script. Makes a lot more sense as you're free to cleanly do anything you'd like there instead of relying on kludges like this.

pindeslandes, 2015-08-31:
Not what I meant.
My point was that I was assuming the list of units available won’t ever be as complete as people may wish, as I have read or heard about people asking to add some more units to the existing list (pressure, energy…).

As I wrote, this feature wouldn’t replace the existing list, but would allow the user to add some other units to it.

In my imagination, it was a win-win situation (more versatility for the user, no more units to wonder if it’s worth to be added for the developers).

kilrah, 2015-08-31:
Yeah I understood that, but it's still very klunky for the user AND complicated to implement for us.

closed 2019-11-15 by stale[bot]