opentx/opentx #2819: Trim of a virtual input not taken into account in Adjust GV special function

pindeslandes, 2015-08-31:
Taranis – OpenTx 2.0.17

I set:
[I1] = Ail x 100% with Trim = YES
Adjust GV1: Source = [I1]
L1 = a>x with V1 = [I1] and V2 = 0

I move the aileron stick to check: GV1 and L1 react accordingly.
Now I move only the aileron trim: L1 still reacts as expected (L1 = 1 when trim > 0) but GV1 doesn’t move at all.

Now, on Companion 2.1.1, I display [I1] on a telemetry screen (I can't with 2.0): [I1] doesn’t move when I move the aileron trim.

kilrah, 2015-08-31:
That's normal, trim is added in the mixer.

Use CH32 = 100% I1, and Adjust GV1 Source CH32 of you want trim.

pindeslandes, 2015-08-31:
I would understand if the trim was added only in the mixer, but it seems to be also added in the LS (see my first comment).

So the real issue would be: ‘Trim of virtual input added in LS’s source’, right?

kilrah, 2015-08-31:
Aha, description didn't really highlight the difference in behavior. 
Then yes one of the 2 is wrong, need to discuss which.