opentx/opentx #2824: Proposal: Companion sync SD should be two way.

rafabene, 2015-09-02:


projectkk2glider, 2015-09-03:
Related to #2743

projectkk2glider, 2015-09-03:
I looked at the sources and looks like it should be two way. What exactly is your problem?

rafabene, 2015-09-03:
The Backup and logs from my sdcard wasn't copied to my computer.

projectkk2glider, 2015-09-03:
@bsongis just one comment. The modification time of a file/directory precision and existence depends on the underlying file-system. If I remember correctly some file-systems have only 2 second resolution, other store file time in UTC. This might be the cause of some issues, but probably not this particular one.

bsongis, 2015-09-03:
If it's true for the backup and logs file, probably no, I will do a check myself later

rafabene, 2015-09-04:
I checked and it's working.. Sorry!

Closing this issue...

closed 2015-09-04 by rafabene