opentx/opentx #2830: Companion 2.1.2N440 doesn't compare models

rafabene, 2015-09-04:
I tried this feature today for the first time with 2.1.2N440 version and it's not working on Windows. The mouse drag and drop doesn't populate any model on the comparing window.

projectkk2glider, 2015-09-04:
Are you sure you dragged the model onto the text and not in a window? As soon as model is dragged correctly the text should change to the name of the dropped model.

fraternl, 2015-09-04:
I'm using this feature quite often.
Just tested if it still works on 2.1.2N440 and it works.
If he never used it before it's quite certain a user error

kilrah, 2015-09-04:
To be fair that way of dragging has never been very intuitive. Could be worth using a more straightforward UI mechanism, e.g. 2 dropdowns that both have the list of models present in the file...

kilrah, 2015-09-06:
Confirmed it works as it always has, edited issue title to reflect what the actual problem is...

rafabene, 2015-09-08:
I can confirm now. It works dragging on the text but it's not intuitive since my first reaction was to send it to the window.

mpaperno, 2017-04-11:
Fixed with #4769 (can now drop anywhere in the window).

closed 2017-04-11 by mpaperno