opentx/opentx #2835: OpenTX 2.1.1, X9E: Telemetry sensors not shon on pages

MarenB, 2015-09-05:
Hi,

I think I found an issue at least concerning Alt (from Vario) and Curr (Vfas).
Both are found in the telemetry menu and they also send data frequently. 
But they are not shown on the telemetry screens, neither as value, nor as bar.

Further, pressing Enter long to jump to a data section (i.e. telemetry) works for the value screen, but nor for the bar screen, so in second case you have to scroll through all possible selections to get to the telemetry sensors.

kilrah, 2015-09-05:
What do you mean "they are not shown on the telemetry screens"?
Of course they aren't until you've added them to a screen. Make sure that if you have multiple sensors with the same name you're using the correct one.

MarenB, 2015-09-05:
Ok, that's strange. I only connected the Vfas once and there's no reason for a second telemetry value named "Curr". However, I was able to choose a wrong one. So I could set it up correctly now. Thanks!

Where are these multiple sensors coming from?

closed 2015-09-05 by MarenB