opentx/opentx #2837: Edge representation is different in print model / OpenTX University / Companion Setup

fraternl, 2015-09-05:
When I look at Edge in Companion the setup page, Companion print model, the radio itself and the explanation on http://open-txu.org/home/continuing-education/logical-switch-functions/ 

I will get different terminology for what is called "instant" on OpenTX university.
In "print model" this is -0.1
IMHO this should be "<<"

It is true I'm not that familiar with Edge.
In this case the explanation on OpenTX university becomes confusing.

Where OpenTX-U speaks of "instant" it is "no release)"  without the opening brace in Companion

fraternl, 2015-09-06:
I made some screenshots giving the 4 representation of the same logical switch
I don't like the "no release)" in Companion 2.1. Maybe it's just me.
It seems to be changed as OpenTX University is giving "instant" for this value.

It would be nice if the "<<" would appear on "print model"
I think it's now giving the value it has internally.

Edge on Print model

[image]

Edge on Companion 2.1   no release)

[image]

Edge on Radio

[image]

Edge on Companion (OpenTX  University)

[image]

vieuxdruide, 2015-09-08:
hi :-) i charged this morning the new open tx 2.1.2
and in companion in french langage 
in the mixages screen the ratio is in english "weight"
[image]
[image]

bsongis, 2015-10-13:
What do we choose? instant everywhere in companion, while we keep << in the firmware as "instant" won't fit?

kilrah, 2015-10-13:
>  instant everywhere in companion, while we keep << in the firmware as "instant" won't fit?

That seems good.

fraternl, 2016-01-08:
We still have 3 representations of Edge
I prefer to have them all the same.

Companion
L2  Edge   SE↓   [0:0(infinite)]
Print Model:
L2  Edge(SE↓, [0:0])
Radio
L2  Edge [0.0:--]

projectkk2glider, 2016-01-09:
I agree, the companion will be changed to show the same as the radio (perhaps with the word explanation included for easier usage).

bsongis, 2016-01-09:
I am not sure it's easy to change this 0 in Qt! It easier to change it in
the radio, but -- is better than 0. Difficult choice...

fraternl, 2016-01-09:
Replace (infinite) with --- in companion
Print model the same as the radio.

Would that work?

bsongis, 2016-01-10:
Not sure it's easy to do in Companion!

fraternl, 2016-01-10:
I don't mean the zero as that seems to be your problem, but if the word "infinite" with the parentheses is replaced by the 3 dashes it would be almost the same as the radio

Albeit with a 0.0 in front with it....

A simple search and replace....  Not?

bsongis, 2016-01-10:
Yes 0 --- is OK, but is it easy to understand?

jtaylor2, 2016-01-10:
No it's not easier to understand.  I'm not sure why there is a request to make them all the same if it means making Companion less understandable.  The radio does not have the resources or screen space to make everything easy to understand, but Companion does so I don't see where dumbing down Companion just to match the radio is an improvement.

fraternl, 2016-01-10:
It is (almost) the same as in the radio, which IMHO is more important than
an extra word which in itself still doesn't explain the behaviour.

Personally I try a lot of things to find out exactly how it behaves and
after I find out I use this knowledge to solve real-world problems.

fraternl, 2016-01-30:
In the nightly I downloaded today I can't see the "<<" in companion when I do a "print model"
Nor does the "<<" come back in the editing screen of logical switches.

This is because it's not merged yet????
Sorry, I still have no dev platform

projectkk2glider, 2016-01-30:
Correct, it is not merged, I have some ideas that still need to be implemented before we merge.

commit 9ae7aea7 (2015-10-13): Fixes #2837
closed 2016-11-27 by fraternl