opentx/opentx #2843: Simulator 2.1.1: In Telemetry screen 13/13 extended list of choice (Long ENT) doesn't work

fred99, 2015-09-07:
[image]

A long ENT on ---- doesn't show the extented list of choice:
[image]

kilrah, 2015-09-07:
More details, it's only on "Bars" that it doesn't work.
On Nums it's OK, albeit requires you to be in edit mode before you long press.

fred99, 2015-09-07:
You are right, it's only on bars in the simulator and in the Taranis V2.1.1, no problem with nums or script.
I was in edit mode (blinking) and able to select the input using + and - buttons so it's a small problem.
Long Ent is so nice so it's immediatly adopted !
My title/qualification is not accurate.

commit e65c131c (2015-09-17): Fixes #2843
closed 2015-09-17 by bsongis