opentx/opentx #2847: companion 2.1.2

vieuxdruide, 2015-09-08:
hi i charged the new open tx 2.1.2 this morning
and when i opened companion (in french)in the mixages screen the ratio is in english weight
[image]
[image]
thanks :-)

kilrah, 2015-09-08:
Actually nearly everything in compare model/print model also comes out untranslated, although translations files are up to date and complete.

vieuxdruide, 2015-09-08:
salut Kilrah excuses moi mais je n'ai pas bien compris la réponse :-( désolé mais mon anglais est bien pauvre
merci

kilrah, 2015-09-08:
J'ai simplement complété la description avec quelques infos supplémentaires pour quiconque va s'y atteler.

vieuxdruide, 2015-09-08:
ok en fait c'est l'ensemble de la page mixage qui est en anglais :-)
je viens de voir ça.

commit abc48390 (2015-09-08): Fixes #2847
closed 2015-09-08 by bsongis