opentx/opentx #2848: 9XR - Timer not working

donvido, 2015-09-08:
With the update to 2.1.2 both of the 9XR's timers aren't working. They simply don't start. A switch triggered countdown worked well but Thr% and Thrs not, although they were correctly shown in the stats window. It seemed they weren't correctly connected to the timer.
I use this hex file:

opentx-9xr-frsky-templates-ppmca-gvars-autosource-autoswitch-battgraph-thrtrace-pgbar-ppmus-en-2.1.2.hex

2.1.1 is working well

kilrah, 2015-09-08:
What do you have as "Throttle source"? A bug was fixed here and might have changed your selection, you'd need to set it back.

donvido, 2015-09-08:
Throttle source is Thr

kilrah, 2015-09-08:
Very strange, can't reproduce and can't see what would have affected that either.
Are you sure of the settings, on the radio itself (no companion)?

donvido, 2015-09-08:
I downgraded to 2.1.1 but I checked all settings of 2.1.2 in the radio a well as in companion. Simulation worked very well.

donvido, 2015-09-08:
ok, forget that with the switch triggered countdown, it also isn't working. I set it up in the radio.
Timer1 RUD 02:00
Minute x
Countdown x

The Screen shows RUD 00:00 and nothing happens when using switch RUD.

closed 2015-09-08 by bsongis

bsongis, 2015-09-08:
Tested on my 9XR

commit 21ac2eca (2015-09-08): Should fix #2848 - Worst case fixes a warning.