opentx/opentx #2852: Bootload Issue

trashmaster9000, 2015-09-08:
Hi,

I have a random Problem powering up my 9XR-PRO:
from time to time it wont boot up, it Shows the Splash Screen but lighter and if i Switch the power Switch to off it wont go off, it takes about 10seconds to go off.
after that i can power it up and it works normal.
FW: OpenTX 2.1.1 & 2.1.2
Bootloader: 1.5

Question:
How can i Update my bootloader?
in the original er9x FW i can do it through the SD-HC menu but under OpenTX 2.1.1 & 2.1.2 i dont have a Option (i can only delete the file)

bsongis, 2015-09-08:
True. For the moment you have to use er9x to update the bootloader and then you may install opentx.

trashmaster9000, 2015-09-09:
Hmm... ok.

but whats with my Problem?
Is this a genaral Issue with the 9XR-PRO, a Hardware Defect or a Software Problem?

Here is a Video:
https://www.youtube.com/watch?v=YzZDyZxJqFU

You can see my Problem at 30 Seconds...

kilrah, 2015-09-09:
I have the same issue once in a while on my AR9X board, so it might be a problem in OpenTX. It is however unrelated to the bootloader as those that run ersky9xr (same bootloader) don't seem to have the issue.

projectkk2glider, 2015-09-19:
Looks like LCD initialization problem or something similar as the back-light is apparently working (judging from your video).

projectkk2glider, 2015-09-19:
@trashmaster9000 Did this issue also happen with OpenTX 2.0.x series or only in the new 2.1.x?

bsongis, 2015-09-19:
Instead I think it's an SD problem!

bsongis, 2015-09-19:
Instead I think it's an SD problem! I would say that the audio task (which is in charge of SD init) doesn't give the hand to the main task. Therefore the splash hangs.

kilrah, 2015-09-19:
Yes that would seem most likely, as (at least for me) after 10 seconds or so the welcome message will play, and I didn't get it to happen without an SD card (didn't test much though)

Tank86, 2015-09-30:
Hi, maybe it's the same issue as #2358, or at least seems to be related ?

biasedlogic, 2015-10-07:
I have this very same problem.
I have not tested a different SD card yet, but I have tested the same HW setup (including the SD card) with ersky9x 217 and 218a..d test versions. The Ersky (MikeB) SW does not have ever an issue booting, while OpenTX 2.1.3 does, especially after Tx is off for a longer time. 
I believe I've never had an issue with a "hot" restart (power cycle) - it boots OK.

SD-Card is the no-name one HK supplies with each 9XR-Pro.

HW: 9XR-Pro + XJT + 3s2100mAh LiPo (the Tx pack from HK) + OpenTX 2.1.3 (I can post exact options later on if necessary).

biasedlogic, 2015-10-09:
It's been around for a while: https://github.com/opentx/opentx/issues/2358

bsongis, 2015-10-20:
Nightly builds are available, would it be possible to have some tests before we release?

biasedlogic, 2015-10-20:
If you provide me with a bin file, i will test it today evening. Got a baby
girl, no time to dig in the net

biasedlogic, 2015-10-20:
OK, I've spent 20 minutes trying to figure out where to get the referenced nightly from.
If someone can post a download link to a bin file, I can test it tomorrow.

bsongis, 2015-10-20:
http://www.open-tx.org/links.html

You will download the latest nightly build for Companion. Inside Companion, you enable the download of nightly builds for both Companion and Firmware, then you will be able you download the latest nightly builds, every day

biasedlogic, 2015-10-21:
Hi,
Issue still persists on 2.1.4N499.
Turnigy 9XR Pro hangs on boot with the opentx-9xrpro-heli-templates-ppmca-gvars-symlimits-autosource-autoswitch-battgraph-bluetooth-ppmus-sqt5font-en-2.1.4N499.bin

Regards,
Marek

bsongis, 2015-10-21:
Does it always hang on boot?

biasedlogic, 2015-10-21:
No, with the latest nightly it booted one time ok and the next time it hanged. The 2.1.3 would boot every now and then without issues, but could also have a streak of five hangs in a row

trashmaster9000, 2015-10-22:
Hi,

Sorry for the late answer...

My 9XR-PRO hangs almost always after i update the eeprom through Companion and almost always on the first start after a Firmware Update.

The bootloader is not affected, if i do a Update, Exit the Bootloader with a Long press on Cancel i get this hang...

blairfancy, 2015-10-25:
Same here still hang on boot.

bsongis, 2015-10-25:
Someone ready to test? Just fire an email at bsongis [at] gmail [dot] com, I will send a bin.

biasedlogic, 2015-10-25:
you've got mail ;-)
M.

biasedlogic, 2015-10-25:
http://jenkins.open-tx.org/issue2852/

still crashes every now and then.

bsongis, 2015-10-25:
do you mean "still hangs on boot"? or "crashes"?

biasedlogic, 2015-10-25:
Sorry, I wasn't precise. 
It still hangs on boot. 

What i have observed is, once it hangs, turning the main switch off doesn't seem to do a lot, the display state doesn't change and the back light stays on. When the power switch is turned on again in this state, the frsky module beeps again, the tx still doesn't start, but now turning the power switch off results in a normal shutdown. 

I can't tell how regularly this can be reproduced though...

bsongis, 2015-10-25:
I just found a bug in Companion, am I right if I say that Companion was not able to re-read EEPROM from a 9XR-PRO?

biasedlogic, 2015-10-25:
Can't tell, but it sure couldn't write the backup back. 
I never trusted companion / eepskye with reading backup on my eeprom, always just copy of by hand.

pfeerick, 2015-10-27:
I can also confirm on a 9XR Pro that this issue exists. I updated from v2.0.3 where I had no startup or shutdown issues (but the graphic 'corruption' on startup due to disabled splash screen existed) to freezing for approx 10 seconds on a cold boot, and usually no startup hangs if I switch the tx off and on again. I have had one or two power off lockups but I'm unsure if low tx battery contributed to the last  time that happened. 

Latest build I've tried is the opentx-9xrpro-heli-templates-nofp-autosource-autoswitch-battgraph-en-2.1.5N508 build. It all seemed good after a first-boot delay hang (which I thought that was partly due to bad eeprom and calibration needed again?). A few off and on cycles didn't bring up any shutdown issues, but pulling the battery and ensuring that the tx was doing a cold boot resulted in a startup hang of about 7-10 seconds again. Get a beep from the FrSky module, and then the delay, and then finally stick warnings or main operating screen. Isn't 100% reproducible though... tried to do it two more times and behaved itself. There do seem to be a few more screen artifacts on the shutdown screen - random lines or blocks lit up for a moment after the tx has shut down, etc. The prior fw (2.1.3 IIRC) seemed to 'fix' the problem by removing the SD card, so definitely seemed related to that. 

Keep up the good work... more than happy to test new builds and provide whatever info I can to help. Sorry I can't give you a way to reproduce it... it is one annoying intermittent gremlin.

VA3WK, 2015-11-09:
i wonder if it has to do with the format of the SD card. I formatted one with a fat table and 32kb and did not have the problem yet. i am using a adata 8GB class 10 card.

pfeerick, 2015-11-10:
I just checked the SD card I'm currently using, and it is FAT formatted 
/w 32k sector size, and is the 2GB cheapie that I got with the 
transmitter. The problem here is that this issue is intermittent (so 
far). I haven't used the transmitter for a couple of days, turned it on 
and started up ok. Turned it off, removed the SD card, started up ok 
(just seemed a tad faster). Turned off, reinserted SD card, hung for 10 
seconds before 'working'. Turned power off, turned back on again, no 
problem. Tried the whole process again, wouldn't hang again. I'll try 
formatting the card and trying again just to be sure though.

VA3WK, 2015-11-10:
I have booth the Taranis and the 9xr pro, have absolutely no problems on the taranis. i switched the 9xr pro from eersky to opentx to have booth on the same system, that's when the problem started.There seems to be a small improvement in the frequency of the delayed startup with a faster SD card. Btw. if you leave it alone it will boot and come up normally.

pfeerick, 2016-02-28:
Whilst I did format the SD card at the time (with just a quick format in Windows IIRC), and that didn't do anything for me, I just downloaded 2.1.7 companion, and wanted to put on the matching firmware, and the bootloader wouldn't let me see the SD card. I removed it, put it in a card reader, copied all the files off, did a quick format using Rufus, and copied the files back on. Viola... files are accessible via the bootloader, and firmware was able to be updated. So perhaps the SD card was corrupt, and also wasn't being properly re-formatted when plugged into the 9XR-Pro. Regardless of the cause, the slow startup issues seem to be resolved for me, and I am now running v2.1.7. Will update again if it rears it's ugly head.

pyr8, 2016-04-24:
I am having the same issue. 9XR Pro will not recognize SD card (tried 3 different cards now).

I was having a "hang on Power off" issue as well, however I have determined that it is due to a sticking power witch, not a software issue. When will the SD card issue be resolved?

kilrah, 2016-04-24:
The issue here is is about the power up blocking/delay, so unless you have that too your problem would be unrelated.

awshima, 2016-06-19:
Same problem here, just got my 9XR-PRO and switched right away to OpenTX.
Sometimes the boot keeps stuck at the OpenTX logo, and switching the power button to the off position tooks some seconds to actually turn off the radio.

moetop, 2016-06-24:
So I have had this issue in the past, and came back to it recently. when I choose to boot ER9X I don't have any issue. When I boot open9x I get a Dim screen and no boot as well as turning off the power switch takes forever for it to shut down. This is after it was working the last time I tried. The only difference is I left the transmitter sit for about a month do the battery was not fully charged. I re-peaked the battery and everything boots. I have 2 radios (9xr and 9xrpro and 2 battery packs. So I tried the other battery pack that had been sitting and Same exact thing.. after full peak charge no issue. 

So my question is.. Does OpenTX Require/check for a certain battery level, that ER9x doesn't during boot. Because I can get this issue to repeat every time with a low battery. Charge it .. no issue.

Dim screen with boot logo, will not complete boot or cycle power and then it will boot (Sometimes). Turn switch off and it doesn't turn off for a very long time (~15 seconds)

scaper28, 2016-07-06:
I have the 9xr Pro 

I am running the latest openTX and I am also having the same boot up freeze , again its random sometimes it will boot up a few times in a row other times it will freeze 3 times in a row.

is there nothing that can be done with OpenTX to fix this issue

NCBob, 2016-09-16:
I have had this problem on two 9xr pros.  The first one I have the hang on boot issue occasionally, but then it will work.

A friend just got a 9xr pro and I'm trying to help him with it and we can't even get it to boot the first time after flashing Open TX either with Companion 2.1.9 , 2.0 or 2.1 or copying the firmware to the SD and installing it on the radio.  Going back to the 9xr pro image and it works fine.

AlexDaniel, 2016-11-19:
Oh, this issue! Yes, I have it too. Annoying indeed, but in my case in rarely hangs two times in a row.

InsanitySquared, 2018-05-28:
I've got the same problem, same circumstances. Works fine without SD card so is definitely a problem with that, as mentioned above. Will keep testing and notify of any remedies found.

aaaaaaudrey, 2020-07-02:
I am currently having this issue on a brand new TX16s. Sometimes, almost randomly, the radio "turns on" when the power button is held, but the screen just stays a dim gray for 30 seconds to a minute, at which time the radio then turns on as normal.

commit aa88651e (2015-10-19): [9XR-PRO] Missing delay in LCD initialization. Will it fix #2852?

I need confirmation from a 9XR-PRO user. I had the bug one time
yesterday but I cannot reproduce ...