opentx/opentx #2854: Haptic vibration issue with new Open TX firmware

GRCT42, 2015-09-08:
After installing the new firmware my radio continues to vibrate when turned on. (Haptic vibration) Not sure what's on the go. Any help would be appreciated.

bsongis, 2015-09-08:
Perhaps you flashed the wrong firmware Taranis / Taranis Plus

GRCT42, 2015-09-08:
Here's a pic of the currently installed firmware version on myTaranis Plus.
[image]

The radio vibrates erratically! not sure if it has to do with loss on signal from the receiver? I have a ezuhf jr module installed with 4ch lite.

I heard somewhere that you now have to create you're own Telemetry alerts? Not sure how to do this.

Thanks for the reply...

GRCT42, 2015-09-08:
Whats this Discover New Sensors?

kilrah, 2015-09-09:
Duplicate of #2722.
And you won't get telemetry out of an EzUHF.

closed 2015-09-09 by kilrah

GRCT42, 2015-09-09:
yeah the same issue as the rest of the guys in that other post... thanks for pointing me in the right direction.

hrenz, 2015-09-09:
there are 4-6 reasons of vibration
some Alarms 
Mid Potis
Timers
Low Batt level
lost of Kalibration values
and so on.