opentx/opentx #2855: Wrong switch files played from sounds/en/model folder.

Mow06, 2015-09-09:
Just loaded 2.1.2 on taranis b01 model.
Found that the wrong files play from the switch positions. 
eg moving Switch C to middle plays file SC-up.wav
Moving Switch C to down plays SC-mid.wav.

Does it in Companion and simulator and on the Tx..
From testing older versions that I had downloaded but didn't get to test I found
that  2.1.2n442 plays correct files - 2.1.2n444 and 2.1.2  play incorrect files.

Mow06, 2015-09-09:
Fixed the physical switches playing, but the LS sound file 
eg L1-off.wav doesn't play when it should.

bsongis, 2015-09-09:
Would you do a check on Logical Switches with my latest commit? Thanks!

kilrah, 2015-09-09:
Hmm someone now reported the name.wav might not be playing anymore in 2.1.3...

bsongis, 2015-09-09:
Strange, switches changes, but notre name!

kilrah, 2015-09-09:
Yeah I said not confirmed because the user isn't one who really knows what he's doing and can't explain things clearly.

I "think" I now understand from a newer post of his that it's OK on power up but that on model change the name plays in the middle or after other sounds instead of as a first thing.

Mow06, 2015-09-10:
Name file is working for me OK.

Mow06, 2015-09-10:
Is this in the nightly build?  2.1.4N450 LS audio files still don't work.

bsongis, 2015-09-10:
Are you sure at least on simu they work (tested with L1-off.wav and L1-on.wav)

Mow06, 2015-09-10:
Just found that LS audio files work on 1 to 9 but not on 10 and above.

bsongis, 2015-09-11:
Stupid me, you are right!

Mow06, 2015-09-11:
Not stupid. I was testing LS12, you were testing LS1. We were both right.

closed 2015-09-11 by bsongis

Mow06, 2015-09-16:
This logical switch above LS9 not palying audio file  problem is still in 2.1.4N456.
Should it be fixed?

3dpinta, 2015-09-17:
I've actually found that LS3 and LS5 are not working on this either. I'm running 2.1.3. It was working just fine in my 2.1.0 and just noticed the change when I updated it last night.

bsongis, 2015-09-17:
The full fix will be in 2.1.4

commit 259a42bf (2015-09-09): Fixes #2855
commit 4571ed24 (2015-09-09): Fixes #2855 - Logical Switches now

Test needed please!
commit 4854ffa6 (2015-09-11): Fixes #2855
commit 33873959 (2015-09-14): Fixes #2855