opentx/opentx #2856: Pots and keys don’t turn Backlight on

pindeslandes, 2015-09-09:
Taranis – OpenTx 2.0.17

When backlight mode is ‘both’, moving any stick or switch will turn the backlight on. But pots and trim and setting keys won’t.

projectkk2glider, 2015-09-19:
As far as pots the answer is the same as in #2857. 

But the keys should turn back-light on if `both` is selected. I have to check that, but I am pretty sure it works.

projectkk2glider, 2015-09-19:
Just tested in `both` mode the sticks, switches and keys trigger back-light. All is OK as far as I am concerned. Please reopen if you feel we missed something.

closed 2015-09-19 by projectkk2glider

pindeslandes, 2015-09-21:
I have checked again:
**Trim keys** don’t trigger the backlight either on Companion simulator or on my Taranis.
**Pots** don’t trigger the backlight either on Companion simulator or on my Taranis and I still find it strange that they souldn’t.
**Setting keys** don’t trigger the backlight on my Taranis **but** they do on the simulator.

pindeslandes, 2015-09-22:
BTW, @projectkk2glider I have no right to reopen an issue I didn't close myself. Please, could you reopen it (if you think it has to be)?