opentx/opentx #2861: Companion: let the LSs’ parameters line always visible

pindeslandes, 2015-09-10:
Th parameters line doesn’t stay visible if you have to use the ‘elevator’ to go down in the list of LSs. Personally I tend to have a hesitation about the last two items when I can’t see them (which one is duration, which one is delay?).

Visible:
[image]

Not visible:
[image]

projectkk2glider, 2015-10-25:
@opentx/commiters An attempt at this issue available in branch https://github.com/opentx/opentx/compare/projectkk2glider/issue_2861_header_always_visible

I am not a Qt guru, so I am stuck with a problem: how to properly re-size table columns width to better show all items? For example `#` column should be narrower, while the `switch` column should be wider, etc...

bsongis, 2015-10-25:
Ok I take it, I don't promise anything!

bsongis, 2015-10-25:
Done. At least it's a little bit better :)

projectkk2glider, 2015-10-26:
How do you like it now?
[image]

bsongis, 2015-10-26:
Nice!

projectkk2glider, 2015-10-26:
I will continue on other pages...

PR #3091 (2015-11-22): Use new TableLayout for Logical switches, Outputs and Special functions
Closes #2861
closed 2015-11-22 by bsongis