opentx/opentx #2862: [Wish] Possibility to constrain Global Variable between 2 values

Sacre100, 2015-09-10:
When special functions "Adjust GVn" are used for in flight setting, it would be great to have the possibility to constrain the GV between 2 values.

For exemple, I use GV to set differential on the ailerons of my gliders. I use the throttle trim to adjust the GV using the adjust GVn special function. Currently, I use 2 logical switch and 2 special function to constrain the GV between 0 and 50. 

My wish is either to have in each phase, the ability to define min and max limit on each GV, or to have special function "Constrain GVn" to do it.

projectkk2glider, 2015-09-19:
Seems like a reasonable request. Bit it will require an EEPROM change in any case so it will have to wait for 2.2.x. 

We need to discuss more and decide on how to implement this.

projectkk2glider, 2015-09-22:
Moved to #2917

closed 2015-09-22 by projectkk2glider