opentx/opentx #2863: [wish] X9E as bluetooth handfree mobile phone device.

Sacre100, 2015-09-10:
Many time, I receive phone call when I am flying on the slope and I'm not able to answer. So I'm obliged to land, to take the phone out of my pocket and check who called me.

As the X9E as bluetooth capability, it would be great to be able to connect the mobile phone to the Tx and when there is a call, to be able to answer just by pressing the ENT key. Something very similar that I have in my car where I can answer to a call by pressing a button on the weehl when my mobile is connected through bluetooth connection.

bsongis, 2015-09-10:
@opentx/commiters Who wants to take this one?

:)

kilrah, 2015-09-10:
Even if we could use it to send the "answer call" command I'm not sure the call quality will be that good with the phone speaker and mic in your pocket :)

closed 2020-04-06 by stale[bot]