opentx/opentx #2867: Play value of consumpt. sensor says "miles per hour" instead of "milliamps per hour"

joaquins, 2015-09-13:
Taranis - OpenTx 2.1.3 (opentx-taranis-noheli-haptic-lua-nojoystick-en)

When a special function is set to play the value of a new calculated sensor that uses the formula "consumpt", the voice says the value followed by "miles per hour", instead the proper "milliamps per hour" or alike.

jtaylor2, 2015-09-13:
Sounds like you are still using the 2.0 sound pack and didn't upgrade it to the 2.1 sound pack when you upgraded your firmware.

From http://www.open-tx.org/downloads.html
Sound Packs

Sound packs can be downloaded either through companion (the Download dialog will point you to the correct directory for the selected radio type), or from the links below:

Sound packs for OpenTX 2.0  http://voices-20.open-tx.org/
Sound packs for OpenTX 2.1  http://voices-21.open-tx.org/

joaquins, 2015-09-13:
Thanks, I reinstalled the sound pack for 2.1 and it's working fine.

closed 2015-09-13 by joaquins