opentx/opentx #2868: Sounds

lvale, 2015-09-13:
Hi,

I had the debug window opened on the simulator and noticed that the sound 0159.wav is being invoked and fails. That specific sound does not appear on

 http://voices-20.open-tx.org/opentx-taranis/en/english-taranis.csv

Didn't check on the radio, only on simulator.

f_close(0x7fff71586070)
playFile("/SOUNDS/en/SYSTEM/0001.wav", flags=0, id=0)
playFile("/SOUNDS/en/SYSTEM/0159.wav", flags=0, id=0)
findTrueFileName(/Users/Luis/Documents/Companion9x/SDCard21/SOUNDS/en/SYSTEM/0001.wav)
	found in map: /Users/Luis/Documents/Companion9x/SDCard21/SOUNDS/en/SYSTEM/0001.wav
f_open(/Users/Luis/Documents/Companion9x/SDCard21/SOUNDS/en/SYSTEM/0001.wav) = 0x7fff71586070
f_close(0x7fff71586070)
findTrueFileName(/Users/Luis/Documents/Companion9x/SDCard21/SOUNDS/en/SYSTEM/0159.wav)
	not found
f_open(/Users/Luis/Documents/Companion9x/SDCard21/SOUNDS/en/SYSTEM/0159.wav) = INVALID_NAME
playFile("/SOUNDS/en/SYSTEM/0002.wav", flags=0, id=0)
playFile("/SOUNDS/en/SYSTEM/0160.wav", flags=0, id=0)

kilrah, 2015-09-13:
> http://voices-20.open-tx.org

You're looking at the csv for 2.0. 
2.1 csv and packs have 0159.wav = "minute".

closed 2015-09-13 by kilrah

lvale, 2015-09-13:
Thanks Kilrah. My mistake for not noticing the 20.

The correct link, for future reference is http://voices-21.open-tx.org