opentx/opentx #2869: Telemetry sensor discovery not working if receiver is paired in D8 mode

joaquins, 2015-09-13:
Taranis - Opentx 2.1.3 (opentx-taranis-noheli-haptic-lua-nojoystick-en)

Paired with FrSky X8R receiver + FCS 40A voltage/amp sensor.

If the pairing is in D16 mode, the telemetry discovery detects correctly the sensors:
- RSSI
- RxBt
- SWR
- Curr
- VFAS

If the pairing is in D8 mode, the telemetry discovery detects only:
- SWR 
- RSSI
- A1
- A2

(all of them as id 25, and no voltage or current are detected)

This same setup in both D16/D8 was working fine in 2.0.x versions

kilrah, 2015-09-13:
When an X8R is bound in D8 mode it does not support any smart port sensors, that's normal. I doubt that was working on 2.0.

joaquins, 2015-09-13:
You are right, sorry for the confusion.

What caused the D8 confusion in my case what to see another receiver in D8 mode, a FrSky D4R-II, also not sending the telemetry from a Naze32.

But fortunately I figured it out: The naze32 ONLY sends telemetry when in ARM state, so the telemetry discovery in the Taranis must be done in this ARM state.

Hope this helps anyone else with the same issue.

Thank you all for the great work and support.

closed 2015-09-13 by joaquins