opentx/opentx #2870: 9XR PRO PPM2 support?

Links2004, 2015-09-13:
Hardware: 9XR PRO
Version: 2.1.4N453

is there a way of enabling the PPM2 port of the 9XR PRO.

may this helps:

[image]

http://openrcforums.com/forum/viewtopic.php?f=122&t=5659

projectkk2glider, 2015-09-19:
@MikeBland you seem to know 9XR-PRO the best. Is this possible?

kilrah, 2015-09-19:
Pretty sure we already have the code in opentx and it's just not enabled for 9xr-pro.

MikeBland, 2015-09-19:
The PPM2 port was always on the SKY board and the 'PRO is basically a copy of that.
I thought PPM2 was already supported, and even open9x seems to have support for PPM2 on the SKY board. It's certainly been in ersky9x for both SKY and 'PRO from when they became available.

bsongis, 2015-09-19:
Yes it's only me, I didn't know PPM2 port was available in 9XR-PRO and I disabled this port. It should only some #define / #ifdef to change.

Links2004, 2015-09-21:
may this one?
https://github.com/opentx/opentx/blob/6c0a9714fc85237af9d17b0b44ee1348c315f08d/radio/src/gui/9X/menu_model_setup.cpp#L704

what is the meaning of the different ```REVX```  and ```AR9X``` for the ```PCBSKY9X```?

MikeBland, 2015-09-21:
There is the original SKY board, then there is the 9XR-PRO (REVX), which is slightly different, and there is a new board, not yet fully for sale, called the AR9X.

commit 1d6adcc2 (2015-09-26): Fixes #2870
closed 2015-09-26 by bsongis