opentx/opentx #2871: Logical switch sounds are not played

lvale, 2015-09-13:
Radio Taranis X9D B, fw 2.1.3

Model name directory created, Switches sounds correctly played if file exists as SA-up.wav and similar. Also name.wav inside model dir plays ok.

But if Logical switches are invoked and file name is L1-on.wav and similar these never get played. Tested all the way to L9-on.wav and L9-off.wav, and yes there are active Logical switches, and we can see their indicator change on the screen, but the sound is not invoked. Also checked on the debug screen and no L*.wav is ever called.

tested both on Companion and on the radio. The wav files are valid and if renamed to SA-dow.wav or similar play normally.

kilrah, 2015-09-13:
Duplicate of #2855 .

closed 2015-09-13 by kilrah

lvale, 2015-09-13:
So, if it's fixed on that commit by Bertrand, I assume it's scheduled for 2.1.4 ?

kilrah, 2015-09-13:
Yep.

lvale, 2015-09-13:
tks

Mow06, 2015-09-14:
Should this be fixed in latest nightly builds.? It's not working in 2.1.4N453. LS 1 to 9 work, above 9 still don't.