opentx/opentx #2872: Voice announcement of timer is odd

zieren, 2015-09-13:
Taranix X9D+, OpenTX 2.1.3 (just flashed from 2.0.15)

The timer is announced in a funny way. I've set Timer1 to TH% 00:04:20. My special function "Play Value: Tmr1" pronounces this as "four point zero and twenty point two". That is somewhat irritating. It used to (and should) say "four minutes and twenty seconds".

Maybe a compact output like "four minutes twenty" would be nice (I for one would like that), but announcing minutes with decimal places followed by seconds is confusing. Subsecond precision is also probably overkill for a timer.

kilrah, 2015-09-13:
You need to update your voice pack to a 2.1-compatible one.

closed 2015-09-13 by kilrah

zieren, 2015-09-14:
Sorry, I missed that on the download page. Works now, thanks!