opentx/opentx #2873: DSM2 receiver model IDs reset on 2.0.15 -> 2.1.3 update, and spurious warnings about conflicts

zieren, 2015-09-13:
Taranix X9D+, OpenTX 2.1.3 (just flashed from 2.0.15)

All my DSM2 receiver model IDs were set to 00 in the update. It would be great if the conversion would copy those. Now I've got to go through all DSM2 models and enter the IDs from the backup.

Also, maybe more interesting: When I set the first ID back to the right value (IIRC 4) I got a warning about "ID in use". But all IDs were zero. Confusingly, when I set another ID back (IIRC 16) I did not get that warning. At first I thought the number of DSM2 models I have programmed is probably >=4 and <16 and that might be related, but after counting them I have >20 (some are defunct).

I still have some at 00, so I can experiment in this state if it's of any help.

kilrah, 2015-09-13:
2.0 only had one ID shared for both internal and external modules, now we have 2 separate ones, and the "old" one is considered to relate to the internal. So it's normal for external ones to be all 0.

The warning is your issue #2187.

closed 2015-09-13 by kilrah

zieren, 2015-09-13:
Why doesn't the conversion check for the case where only the external module is enabled, and in that case copy the ID to the new dedicated external ID?

Can you explain why the warning is shown even though the ID that was just set isn't duplicate?

I don't understand what this issue has to do with #2187, which is about internal copying.

kilrah, 2015-09-13:
> Why doesn't the conversion check for the case where only the external module is enabled, and in that case copy the ID to the new dedicated external ID?

Because nobody decided to code that in - probably because users of the DSM modules are a small minority and it wasn't deemed worth the effort.

> I don't understand what this issue has to do with #2187, which is about internal copying.

See my post in the other thread:

>There are also other adjustments needed... When checkModelId() was written there was only one ID for both modules, now we have separate ones for both. I've noticed the warning comes up even when a module is disabled, which makes little sense. Might be better to skip the check if the module is set in a mode that doesn't use the ID, and rather do the check when said module is enabled.

Currently the check runs over both modules even if one is disabled. The warning you get is because the ID for the (disabled) internal module is duplicated, and the need for changing that is expressed in that post.

zieren, 2015-09-14:
I see, thanks for the explanation.

griebd, 2015-09-17:
I will comment in #2187 and work there.
BUT, the warning about duplicated ids in 2.1 wasn't done by me on that issue... I did change only 2.0.
The alert was already implemented, but apparently there is a bug! It does not take in account the separated ids for external and internal modules. (at leas on the eeprom converted from 2.0)