opentx/opentx #2874: Can't flash bootloader from inside OpenTX

zieren, 2015-09-13:
Taranis X9D+, OpenTX 2.1.3 and 2.0.15

I put the firmware file, opentx-taranisplus-lua-en-2.1.3.bin, in the FIRMWARES directory (no other files in there). But the SD card browser in the radio setup menu shows an empty directory. Files in other directories are shown. This happened on 2.0.15 and is still happening with 2.1.3.

Am I doing something wrong? Shouldn't it give some indication as to why it's hiding this file? It's a bit frustrating to just not see it.

fred99, 2015-09-13:
The total number of character is 32 including the extension file name. So max length is 28 and “.bin” = 4 (dot is counted as 1 character).

kilrah, 2015-09-13:
Correct. I doubt that will change, but let's get others devs' input.

zieren, 2015-09-14:
I wondered if that was the cause, but I was reluctant to rename a file when the name was obviously having some effect. Since the filename encodes firmware features I was concerned it might confuse OpenTX if I just renamed it to foo.bin...

The 32 character size limit is not for the file system, right? Maybe rather than hiding those files they could be shown multi line, or even truncated. In most cases there will only be one file in there anyway.

projectkk2glider, 2015-09-19:
>Since the filename encodes firmware features I was concerned it might confuse OpenTX if I just renamed it to foo.bin...

No the name has no meaning to the OpenTX as long as it has `.BIN` extension.

>The 32 character size limit is not for the file system, right? 

The limit is there to conserve memory. We could easilly expand it, but other features would suffer (in this case mostly RAM for Lua scripts).

>Maybe rather than hiding those files they could be shown multi line, or even truncated. In most cases there will only be one file in there anyway.

You are making dangerous assumptions there. What if there are more files with long filenames that have first 32 characters the same? Perhaps multi-line display would be possible with some code gymnastics, but it could get ugly pretty fast. @bsongis what do you think about multi-line file name display?

zieren, 2015-09-19:
I realize that multi-line display might be too much effort for too little gain. But I imagine that the 95% use case would be covered with a logic like this: If there is only one file, show the file name truncated. If there are >1 files, show a message that says to remove all but the one that is to be flashed.

Would it be an option to only show the full filename once the file has been selected?

griebd, 2015-09-19:
I don't know how it's implemented now, so my suggestion may be wrong...
But, just to make life easier to the average user, we could have a pop-up warning when a folder with file names too long is opened. Just say something like "Some files are hidden because their file name are too long."

zieren, 2016-09-05:
Friendly ping :-) 

I noticed that there are video instructions around that don't mention this and thus might be confusing (https://www.youtube.com/watch?v=TPZvhlzD6rc).

powercycle, 2017-03-06:
While this is being decided, I mentioned this in the flash bootloader section of the advanced settings section in the manual.

When I get to updating the companion portion, I'll do it there as well.

closed 2019-11-15 by stale[bot]