opentx/opentx #2880: PPM centre does not work with the DSM hack module.

pmackenzie, 2015-09-15:
I noticed quite a while ago that in 2.0.13 adjusting PPM centre had no affect on the servo outputs.
I didn't report it because I am running a fairly old version of opentx.
Another RCG user has reported the same thing in 2.1.2, so perhaps it is time to look into it?

He postedd here:
http://www.rcgroups.com/forums/showpost.php?p=32656728&postcount=1100

Then added that he was in fact using the hack module here:

http://www.rcgroups.com/forums/showpost.php?p=32690056&postcount=1185

Thanks

Pat MacKenzie

mcddd, 2015-09-17:
I am the person that Pat referred to in his post.

The PPM center not working only seems to apply to using the DSM2/X hack module.  The same transmitter using an frsky D6FR receiver and the internal RF allows the PPM center adjustment to work normally.

Here is a link to a youtube video of the problem:
https://www.youtube.com/watch?v=YTQT__67zhc

Thanks,
mcddd

bsongis, 2015-09-18:
Does this DSM2/X module accept DSM2 protocol or PPM as input? Perhaps I forgot to take care of PPM center in DSM2 encoder, that's very possible

bsongis, 2015-09-18:
I just checked and yes I forgot it in DSM2 encoding. I will fix this later today so that you can check tomorrow morning in the nightly builds.

pmackenzie, 2015-09-18:
HI, any chance this could be done for 2.0.X as a bug fix?

Thanks,

Pat MacKenzie

projectkk2glider, 2015-09-19:
>HI, any chance this could be done for 2.0.X as a bug fix?

Yes.

bsongis, 2015-09-19:
Yes I will take care of it.

@projectkk2glider @kilrah I think the PPM_CENTER is not correctly added in pxx_arm.cpp, would you double check? If I am not mistaken PPM_CENTER unit is in us, it should be added to channelOutputs after being multiplied by 2 ... and the ratio 512 / 682 should be applied (it is not in the current 'master'). If I am right it should be fixed but in 2.2.0, we cannot change this in a minor release.

projectkk2glider, 2015-09-19:
>If I am not mistaken PPM_CENTER unit is in us

Correct, the default value is 1500.

Current math for PPX is:
```
chan = PPM_CH_CENTER(i) - PPM_CENTER + (channelOutputs[i] * 512 / 682) + 1024
```

And this is used for display:
```
    lcd_outdezAtt(13*FW, 0, PPM_CH_CENTER(sub)+channelOutputs[sub]/2, 0);
```
Which suggests that the correct math would be:
```
( (PPM_CH_CENTER(i) - PPM_CENTER) * 2 + channelOutputs[i]) * 512 / 682) + 1024
```

commit 1c16a99d (2015-09-20): Merge pull request #2900 from opentx/bsongis/issue2880_2.0_ppm_center_for_dsm2

Fixes #2880
commit 4040828e (2015-09-20): Fixes #2880
commit 1aa51436 (2015-09-20): Merge pull request #2899 from opentx/bsongis/issue2880_ppm_center_for_dsm2

Fixes #2880 - PPM_CENTER should be ok now with DSM2 pulses
commit 6389f977 (2015-09-20): Fixes #2880 - PPM_CENTER should be ok now with DSM2 pulses
closed 2015-09-20 by bsongis