opentx/opentx #2881: Bootloader version did not update

shadowjig, 2015-09-15:
I updated from 2.0.15 to 2.1.3.  The firmware updated fine and the eeprom converted.  However the bootloader still shows and old version of 2.0.12.

fraternl, 2015-09-15:
That's normal behaviour and has been this way all the time.
If you want the bootloader to get updated you should not flash it when it's in bootloader.

You should turn the radio off without USB attached.
Then connect USB cable and write the firmware.

User error

kilrah, 2015-09-15:
The bootloader is not supposed need updates, so if you want to update it you have to do it manually.

closed 2015-09-15 by kilrah

fraternl, 2015-09-15:
Can't the bootloader get its own version number so this is expressed as well?
That's what version numbers are for.