opentx/opentx #2882: OpenTx 2.1.4N455 Delete all sensors

flarssen, 2015-09-15:
1. If the idea is to get rid of sensors discovered by accident, I would suggest to rename it to Delete sensors, then have three selections in the popup: Discovered, All, Cancel. This would make it possible to preserve added sensors.

2. Seems that reference to sensors in telemetry pages and logical switches is by index. If all sensors are deleted, and the list is populated again, the reference might be to a different sensor or non at all. 
Protect referenced sensors, maybe?

fraternl, 2015-09-15:
I don't think openTX "knows" of a sensor is added or not. I'm not sure of this.

I'm in favour of a feature that optionally protects all referenced sensors.

flarssen, 2015-09-15:
Right. Didn't realize discovered sensors were of type "Custom" too. But it can differ between "Calculated" and "Custom" ones. Anyway, if one add a sensor, it will probably be referenced, so protecting these would solve the problem.

projectkk2glider, 2015-09-19:
So we are left with the options:
* ALL
* UNUSED - sensors that are not referenced anywhere.

But sensor usage/reference can not be checked for the Lua scripts! So the UNUSED option is out also. That leaves us with the current state.

closed 2015-09-19 by projectkk2glider

kilrah, 2015-09-19:
This is meant for testing setups more than anything else, or maybe very early in your model setup to get your sensor list right before you've even started programming LSs and the like.