opentx/opentx #2884: Lua i/o model in 2.1.3

wolkstein, 2015-09-16:
hi,
simple question, is lua i/o model supported in 2.1.3? and if yes, how to use it in a way as opentx developers create the implementation.
i mean sd card structure(user space). maybe callback functions to noticed shutdown  and so on.
an example would be helpfully:).

thx
wolke

bsongis, 2015-09-17:
It seems to me that I added it, yes. Did you try the standard i/o functions of the Lua manual?

wolkstein, 2015-09-18:
not under 2.3.1. at least i test it under 2.0.17 without success. and than i use it as described in this reference. http://www.lua.org/pil/21.2.html

projectkk2glider, 2015-09-18:
Original issue was #1471 and these are implemented IO functions:
```c
/*
** functions for 'io' library
*/
const luaL_Reg iolib[] = {
  {"close", io_close},
  // {"flush", io_flush},
  // {"input", io_input},
  // {"lines", io_lines},
  {"seek", io_seek},
  {"open", io_open},
  // {"output", io_output},
  // {"popen", io_popen},
  {"read", io_read},
  // {"tmpfile", io_tmpfile},
  // {"type", io_type},
  {"write", io_write},
  {NULL, NULL}
};
```

projectkk2glider, 2015-09-18:
@wolkstein reopen this issue if it is still not working for you (after you study my comment above). You should try it first in the Companion simualtor and use the Debug window (F6) and add some `print()` statements to your Lua script.

closed 2015-09-18 by projectkk2glider

kilrah, 2015-09-18:
> not under 2.3.1. at least i test it under 2.0.17 without success.

It is only available in 2.1.