opentx/opentx #2891: Selectable precision for current ID 0x28

flarssen, 2015-09-17:
Ref. #2607 
I have a sensor with 20mA resolution, so having a similar solution for ID 0x28 would be great. Would also  be useful when creating sensors for low current measurements, for Rx battery etc.
Maybe use 5000 instead of 2000, so 0-500A with one decimal and 0-605A with two decimals.

projectkk2glider, 2015-09-19:
Both D and Sport sensors have current resolution of 0.1A. So any change would have to be done for both types. 

The change in #2607 was simple since the stored voltage was already in high precision mode (two decimal places). This is not the case for current. The current is stored with one decimal place now. So we would have to change that also.

I wonder if this is really smart thing to do? Will we gain anything from it? Will the consumption be accurate enough for small capacity batteries for small currents?

flarssen, 2015-09-19:
I tried using ratio as a workaround, but this works strangely for current. For temperature, I can send it in 1/10 degrees, specify a ratio of 25.5 and get temperature with one decimal. For current, a ratio of 25.5 gives me the original value. To get a division by 10 I would have to set ratio to 2.55, which is not possible since ratio is one decimal.

projectkk2glider, 2015-09-19:
The ratio needs improvement indeed, see #2615

flarssen, 2015-09-19:
Thanks. Then I wait for a working ratio. Probably the correct solution for such cases.

closed 2020-07-07 by stale[bot]