opentx/opentx #2894: Feature request: Selectable serial outputs instead of PPM

DTFUHF, 2015-09-18:
Expanding on the DSM2/DSMX serial protocol that is already available but limited to 6 channels, I'd like to see some serial protocols available on the PPM output pin in the module bay. Specifically SUMD, but also SBUS, expanded Spektrum, etc. would ensure a broad range of compatibility with radio modules that accept a serial input. PXX is difficult to decode on lower end hardware (AVR) and has not seen widespread acceptance. PPM latency is high vs. these serial protocols, especially on 16 channels.

fujin, 2015-09-18:
++++:100:

mlyle, 2015-09-18:
Or just turn up the DSM to use more channels. :P

DTFUHF, 2015-09-18:
I'm not able to do this, it's limited to 6 channels.

mlyle, 2015-09-18:
@dtfuhf -- not sure I understand.  http://wiki.paparazziuav.org/wiki/DSM  The actual protocol appears capable of carrying 18 channel streams.

DTFUHF, 2015-09-18:
Yes, but I'm unable to do this in OpenTX.

bsongis, 2015-09-18:
Why not SBUS on the PPM Pin.
Or perhaps we could use the S.PORT Pin for sending pulses in whatever serial format (there is an hardware serial on this pin), and Heartbeat Pin for receiving Telemetry. 
Because bitbanging on the PPM Pin uses CPU ...

DTFUHF, 2015-09-18:
I would definitely be open to any form of serial TX on the module bay pins.

kilrah, 2015-09-18:
The only 2 pins we can drive in there are pin 1 (usually PPM), which in terms of hardware driving possibilities is a timer output, and pin 5 (smart port), multiplexed, inverted serial port.

On pin 2 we can receive only, inverted serial port.

MikeBland, 2015-09-18:
Is this related to this project?: http://www.rcgroups.com/forums/showthread.php?t=2165676.
Er9x and ersky9x (including on the Taranis) are supporting this serial protocol, on the PPM pin. It is a variant of the DSM hack protocol.
Also, for serial receive, on ersky9x I have "bit-bang" code working to receive serial data, both inverted and non-inverted, on a processor pin. This works by making sure all other interrupts are a lower priority, then using an "interrupt on pin change" interrupt at the highest priority to detect the edges on the serial input and decode them into a serial data stream.

phobos-, 2015-09-20:
Seeing so many Taulabs guys I'm guessing it's for Taulink module, or OpenLRS. IMO adding SBUS or 115200+ baud Serial protocol on pin 5 transmitting all 16 channels with some kind of interrupt similar to XJT heartbeat would be the best and quickest approach with least latency.
@DTFUHF why you need it specifically on the PPM pin, I think you already have telemetry serial port wired to pin5 (TX) and pin2 (RX), why not use that instead? At least in your deluxe tx.

DTFUHF, 2015-09-20:
TauLabs guys are here since we were discussing this issue in #taulabs on freenode IRC :)

I don't require any specific pin, the functionality itself benefits more people than just me. But I do have pins 1 and 2 connected to RX on some hardware.

bsongis, 2015-09-24:
[image]

bsongis, 2015-09-24:
For the moment 16 channels sent every 5 ms with SBUS protocol on S.PORT Pin. But careful it's inverted serial!

mlyle, 2015-09-24:
@bsongis Nice!

DTFUHF, 2015-09-24:
Very nice, thank you :)

Nub, 2015-09-24:
Awesome! Can't wait to give her a test!

fujin, 2015-09-24:
This is soo sooo sooo rad

DTFUHF, 2015-09-24:
This is working for me, here is a build for Taranis, with PPM_UNIT=US HAPTIC=YES

https://drive.google.com/file/d/0B3BIhKzLAI86ZXRaWmFld3RxVG8/view?usp=sharing

Machtap, 2015-09-25:
Getting firmware incompatible with the Taranis+ download, can anyone smarter than me test it to confirm and/or compile a taranis+ binary with this change?

DTFUHF, 2015-09-25:
Yeah sorry about that, I'm not sure how to compile for taranis+.

bsongis, 2015-09-25:
make PCB=TARANIS PCBREV=REVPLUS

DTFUHF, 2015-09-25:
Hmm, that's what I've done, and that's the second person reporting it isn't working. Sorry but I don't have a Taranis+ to check.

kilrah, 2015-09-25:
That's again a non-plus firmware.

DTFUHF, 2015-09-25:
OK, one last time, at the link above, did make clean before building again.

kilrah, 2015-09-25:
Still the same it seems, but as you're always using the same filename I'm not sure I got the latest version.
Note that the make arguments must be as shown i.e. in caps and without spaces before/after the "=".

DTFUHF, 2015-09-25:
Renamed and reshared to see if it's simply expecting google drive to work like dropbox: https://drive.google.com/file/d/0B3BIhKzLAI86QmUwY2dkYzVDNWs/view?usp=sharing

Machtap, 2015-09-25:
This latest build worked great!  You are all marvelous people!

acc-, 2015-11-22:
What was the reason to replace SBUS with Crossfire protocol ? And which one is marked for next 2.1.X release, Crossfire ? Why not having a choice for two of them, once SBUS was already implemented?

The reason I ask about SBUS is that OpenLRSng TX can accept SBUS input without any modifications - giving full digital processing by TX/RX modules.

bsongis, 2015-11-22:
We would like to have a bidirectional link between the TX and the module on the S.PORT pin. At 400k bauds

acc-, 2015-11-23:
Do you plan to send anything else to TX module except stick positions (thefore making the output frames not sbus compatible), or just willing to receive from TX ?

BTW, can it be 250k or 500k insteadof 400k ? It will allow AVR based TXs to communicate without any errors. Atmega328P at 16Mhz (typical for OpenLRSng TXs) will have 20% of errors at 400k according to AVR's UBRR register rules, but 0% of errors at 250k or 500k.

DTFUHF, 2015-11-23:
Is there documentation on this 2way protocol? It would be great to add to OpenLRSng as well.

DTFUHF, 2015-11-23:
Does it just send channel values? Or is there room for model match as well? I guess this would be explained in the documentation :)

Is there a reason for 400k? We've routinely used 921600 on STM32 targets, if high speed and low latency are the goals of this protocol.

DTFUHF, 2015-11-23:
Or, as @acc- stated, 500k or 1.0M baud would ensure compatibility with AVRs @ 16MHz

kh4, 2015-11-23:
ATMEGA328p can achieve 400 000 bauds. U2Xx=1 and UBRx=4

acc-, 2015-11-23:
Right, forgot about double speed flag...

So now it's just a matter if this crossfire protocol will always output sbus-compatible frames to TX. If it won't change, all we need at OpenLRSng side is to support 400k (rather as an additional serial input for TX). Plus maybe crc checking ?

kh4, 2015-11-23:
To be honest I kinda hate the SBUS protocol, the 11 bit stuffing et all is ankward and the SBUS is not solidly documented either. Something like SUMD would be far better.

kh4, 2015-11-23:
Oh and not forgetting the ultra stupid Futaba only scaling of the PPM values...

bsongis, 2015-11-24:
Things are not finished on crossfire side, I would like they do a test at 400k (and check we are able to decode a serial telemetry stream at this speed) before releasing anything.

@kh4 I choosed SBUS because I had the code ready, but I am ok with everything, I don't like their bits order as well.

In all cases we can add a dedicated protocol for the DTFUHF if you need!

DTFUHF, 2015-11-24:
Let's not forget: https://xkcd.com/927/

If the Crossfire protocol is still being developed it'd be great to submit some suggestions now, so the protocol isn't quickly obsoleted, too cumbersome to use, or suffers when new versions of the protocol break compatibility.

bsongis, 2015-11-24:
Yes go on :)

kh4, 2015-12-26:
Kinda sad for crossfire being inverted serial, I guess the blame is on the Taranis hardware design being tailored for FrSky requirements.

I also would like to see it doing 12 bits instead of 11, the major benefit being able to decode 2 channels from 3 bytes where as decoding 11 bit channels means decoding 8 chs from 11 bytes. The lost bandwidth would be totally negligible even at 200 000 bauds.

mikeller, 2015-12-26:
It should be possible to make an option available to receive an 'inverted' telemetry signal, even for the Taranis. I recently added this feature to the 9XR pro code in OpenTX.

kilrah, 2015-12-27:
> Kinda sad for crossfire being inverted serial, I guess the blame is on the Taranis hardware design being tailored for FrSky requirements.

No, the blame is on the boards that implement inverted SBUS...

bsongis, 2015-12-27:
The CROSSFIRE is inverted because it takes a lot less CPU on Taranis, we would have to do the bitbanging ...

Now why wouldn't we add a new protocol with 12bits etc. We have the possibility to do exactly what we want!

ChrisPrefect, 2016-01-31:
This is a very nice feature! Any chance to add the serial protocol to the trainer port too, so the crossfire protocol can be used with the Taranis 9XE?

kilrah, 2016-01-31:
The trainer signal isn't wired to a serial port, so no. But you can use itthe same way by connecting to the module port.

ChrisPrefect, 2016-01-31:
I did not find a module port on the 9XE. Only a trainer port.

kilrah, 2016-01-31:
An additional module can be installed inside the radio, the port is there.

fiveangle, 2016-04-05:
so it appears there is proof-of-concept code out there, so what obstacles must be overcome in order to get this over the goal line ?

- [ ] identify owner of new feature on OpenTX side
- [ ] identify owner of new feature on OpenLRS side - likely DTFUHF (kha)
- [ ] both sides to agree on what low-latency serial protocol to support that will not be limited to 6 channels as the current DSM2/DSMX implementation
- [ ] other ?

geekness, 2016-08-28:
Sorry to bring up a somewhat old thread, but just wanted some clarification on whether Im using this correctly.
Im using OpenTx 2.0.17 (2015-05-04), and OpenLRSNG (on Delux JR Module) Version 3.8.8
My Receiver is the RFM22b module that comes with Sparky 2.

Am I able to communicate by SUMD/SBUS or some kind of low latency serial protocol?
How?

kilrah, 2016-08-28:
Basically, no. 

OpenTX 2.1 supports crossfire out, and 2.2 will support MULTI out in addition (a slightly modified SBUS), but AFAIK neither of these are supported as input in OpenLRSng.

DTFUHF, 2016-08-28:
Is there a reason why SBUS was removed from openTX? This was working fine with OpenLRSng.

kilrah, 2016-08-28:
It has never been in.

bsongis, 2016-08-28:
@DTFUHF if you have your own protocol don't hesitate to tell me, I have one module for testing. SBUS is the base of CROSSFIRE, but we extended it so that we can send data in addition to channel values, and receive telemetry (everything through the same S.PORT PIN)

dustin, 2016-08-28:
Sbus2?

schwabe, 2016-08-28:
No, a custom protocol that is similar to Sbus on Remote to module.

geekness, 2016-08-29:
Having a "DTFUHF" protocol would be ideal.
We really just need to find the fastest way to get the info to the receiver, however that is.

mlyle, 2016-08-29:
OpenLRS and "fast" are kinda mutually contradictory.

While I agree that shaving off some of the PPM latency is desirable, the over-the-air update rates for OpenLRS are low for link budget reasons, and difficult to make synchronous to updates from the transmitter.  Updating the module at 60Hz or 600Hz is going to have the same effect.

kh4, 2016-08-29:
I think the main point for a new protocol would be to include control and telemetry into one bidirectional data stream.

For latency reasons the RC info could be sent from the mixer like 100Hz the radio module can then use the latest it has when sending out an RF packet. It would be possible to also include a special frame that can be used to synchronize opentx mixer, basically a packet that is sent a fixed time before next RF frame is going to be prepared so that the opentx can run mixer and send out up to date data just before the deadline expires.

The bidirectional link can run at fairly high speed maybe at something like 1Mbps.

At one point I gave this a little thought but got sidetracked

https://github.com/openLRSng/openLRSngWiki/wiki/OpenRcBus

kilrah, 2016-08-29:
> The bidirectional link can run at fairly high speed maybe at something like 1Mbps.

But can the AVR implementations cope with that?

Crossfire is fast and bidirectional, but does require some grunt to process. That's why the multi went with separate lines at the 100k SBUS baudrate.

>It would be possible to also include a special frame that can be used to synchronize opentx mixer, basically a packet that is sent a fixed time before next RF frame is going to be prepared so that the opentx can run mixer and send new up to date data just before the deadline expires.

That's all existing, just not on a single line. The standard with the XJT is control data sent out on PPM pin (1), telemetry and uplink data on SPORT pin (5), and sync on pin 2 (not used at this point, but I believe ersky9x has code that could be reused).

[129 more replies]

closed 2016-12-22 by schwabe