opentx/opentx #2895: Feature request: Multilanguage simulator for Companion and Firmware simulator 2.1

fred99, 2015-09-18:
Under W10, I have installed a functionnal build OpenTX environement on my PC.
I am able to produce the simulator 2.1 in french, using the implemented directive of compilation 
cmake . -DTRANSLATIONS=FR.
I share localy this translation on http://frskytaranis.forumactif.org/t2536-compilation-de-companion-2-0-et-2-1-et-du-simulateur-en-fr using my Dropbox (I don't provide Companion, it's not safe for me, my environement is not OpenTx certified !)
The goal was to be able to make a french doc with nice snapshots from the PC.

Pushed by the forum, I have done a step further (thanks to Kilrah for 2 or 3 key advises) , and I am able to produce the 11*12=132 DLL opentx-xxx with this simple cmd file:
build-all.cmd
Code:
    del CMakeCache.txt
    cmake -G "Visual Studio 10" ..\src
    cmake --build . --target ALL_BUILD --config Release
    FOR  %%L IN (FR,IT,DE,ES,CS,FI,HE,NL,PL,PT,RU,SV) DO (
    cmake . -DTRANSLATIONS=%%L
    cmake --build . --target ALL_BUILD --config Release )
At this step, I don't know if this proof of concept is enough to have a Multilanguage simulator implemented in a next version of OpenTX?
Few questions:
If we have an installator able to select the DLL for one language, there is nothing to change in Companion and the simulator but we loose the possibility to come back to english to document a problem in Gihub.
If we keep all DLL, in the Simulator, we will have 1/132 to select in the dropdown box.

fred99, 2015-09-23:
Hi guys!
Nobody to comment a nice feature with small effort !
If that don't make sense, I have no problem to close it.

projectkk2glider, 2015-09-23:
I am afraid it is not that simple. We have to make modifications to the existing cmake configuration files in order to have the new simulator libraries in the installation packgage, Then the Companion must be modified to run appropriate simulator variant (based on selected language). It might get done eventually, but for me it is not a high priority, lots of other issues still open.

fred99, 2015-09-23:
Ok, no problem, I can see the number of openned issue and I can only thank you, and make some donations.
Should I close it ?

projectkk2glider, 2015-09-23:
No, leave it open.

kilrah, 2015-09-23:
The windows installer generation at least automatically finds and adds all *-simulator.dll files, nothing to do. 
For companion to load the one that matches the language selected in the settings yes there is work. But even without this at least the standalone simu can make use of them.

But yes I don't like the batch solution so much, the generation of all languages should ideally be managed by the build environment.

fred99, 2015-09-23:
I agree, I am able to use cmake, but not to program it, hence this implementation done.
This is more a proof of concept !

bsongis, 2015-09-23:
Yes I implemented this isolation between Companion and the simulators which now run in dynamic libraries. It remains:
[ ] update the compilation servers so that they are able to compile / serve these libraries just as we do for the firmwares
[ ] implement in Companion the mechanism to download libraries
I will take point 2, but if you could take point 1, it would be great!

fred99, 2015-09-23:
You talk to me ? "replique célébre de Robert de Niro"
or to the devs team ?

bsongis, 2015-09-23:
Yes to you :)
Or any dev willing to help!

fred99, 2015-09-23:
Ok, I take the challenge but not sure to success.
I am not really a dev but more a hacker but on the french forum, I can have the help of Sacre100 who burns to code.
And I have an operationnal mvc++ env and know how to use the debugger.

fred99, 2015-09-23:
You write "I will take point 2, but if you could take point 1, it would be great!"
Isn't it the contrary ?
1= update the compilation servers so that they are able to compile / serve these libraries just as we do for the firmwares
It's on your side ?

bsongis, 2015-09-23:
Comment updated

fred99, 2015-09-23:
OK, I agree to take in charge  "implement in Companion the mechanism to download libraries", It's a challenge, I am not sure to success I am more a hacker than a dev.
For companion to load the one that matches the language selected in the settings:
[image]

fred99, 2015-09-23:
To far from my level of expertise :(
return GetFirmware(id);
+		data	0x06209de2 "opentx-taranis-haptic-timer3-noheli-lua-massstorage-mixersmon-sqt5font-fr"	unsigned short *
I see the "fr" at the end but I don't know where those data are loaded/stored (windows register, eeprom)?

bsongis, 2015-09-23:
I am sure all hackers are able to understand this string construction for a firmware :) Have a look how it is done for downloading a firmware!

fred99, 2015-09-23:
Sure, no problem with the concatenation.
All elements collected via the QT interfaces are stored somewhere.
When I run Companion, all the configuration is loaded.
When select "modifier les réglages", the language is there. 
So if I can access to this value, I have just to add this when companion download the simu

SimulatorInterface *GetCurrentFirmwareSimulator()
{
  QString firmwareId = GetCurrentFirmware()->getId();
  QString localFirmwareId = firmwareId + QChar('-') + g.locale().left(2);
  SimulatorFactory *factory = getSimulatorFactory(localFirmwareId);
  if (!factory)
    SimulatorFactory *factory = getSimulatorFactory(firmwareId);
  if (factory)
    return factory->create();
  else
    return NULL;
}

fred99, 2015-09-24:
Sacre100 gives me a clue, the code to create the name is in eeprominterface.cpp

fred99, 2015-09-24:
The language is simply defined here : {
languages.push_back(lang);
}
and lang = "en"
It's a piece of cake if we have the selected language value:)

fred99, 2015-09-25:
With the help of Sacre100 and the debug mode, I saw how it is done:
[image]

The while loop is looking for 
"opentx-taranis-haptic-timer3-noheli-lua-massstorage-mixersmon-sqt5font-fr"
"opentx-taranis-haptic-timer3-noheli-lua-massstorage-mixersmon-sqt5font"
"opentx-taranis-haptic-timer3-noheli-lua-massstorage-mixersmon"
.
.
"opentx-taranis"
Hence at the 2nd loop, we loose the language.
At least, if we just keep it, we have the solution to load opentx-taranis-fr, the prefixe seach name and then to load opentx-taranis-fr-simulator.dll

We have a qDebug() << "searching" << simuName << "simulator";
Is there a way to activate a debug console in Companion ?

Sacre100, 2015-09-26:
With Fred, we have changed and tested getSimulatorFactory in simulatorinterface.cpp.

With this change in companion.exe, for example, if the dll "opentx-taranis-fr-simulator.dll" existe in the directory, the simulator use it instead of the usual "opentx-taranis-simulator.dll". If the dll does not exist the default dll is used.

In simulator.exe, if there is dll in many language, simulator.exe propose them and the user can select the firmware and the language he wants.

We have test this change in version 2.1.3 and the behavior of companion.exe and simulator.exe is as expected.

    SimulatorFactory *getSimulatorFactory(const QString &name)
    {
      QString simuName = name;

    // PREMIERE boucle avec le code langue
      QString simuLang = simuName.right(2);
      simuName = simuName.left(simuName.length() - 3);
      while(1) {
        qDebug() << "searching" << simuName << simuLang << "simulator";
        foreach (QString name, registered_simulators.keys()) {
          if (name.contains(simuName + '-' + simuLang)) {
            simuName = name;
            return registered_simulators[simuName];
          }
        }
        int pos = simuName.lastIndexOf('-');
        if (pos <= 0)
          break;
        simuName = simuName.mid(0, pos);
      }

    // DEUXIEME boucle sans le code langue
      simuName = name;
      while(1) {
        qDebug() << "searching" << simuName << "simulator";
        foreach (QString name, registered_simulators.keys()) {
          if (name.contains(simuName)) {
            simuName = name;
            return registered_simulators[simuName];
          }
        }
        int pos = simuName.lastIndexOf('-');
        if (pos <= 0)
          break;
        simuName = simuName.mid(0, pos);
      }
      return NULL;
    }

bsongis, 2015-09-26:
Why not something which would check all combinaisons and take the best one?

fred99, 2015-09-27:
Another version, this time with only one loop, it's better but not generic. With opentx-taranis-simulator.dll renamed opentx-taranis-en-simulator.dll, the code would be more generic and simple ;)
If I test only simuLang.contains("en"), I am not able to cover all cases between companion and simulator.
[image]

fred99, 2015-09-27:
5 spaces to get a nice presentation !
		SimulatorFactory *getSimulatorFactory(const QString &name)
		{
		  QString simuName = name;
		  bool notEN=false;
		  QString simuLang = simuName.right(2);
		  if ((simuLang.contains("de"))||(simuLang.contains("es"))||(simuLang.contains("fr"))||
			  (simuLang.contains("it"))||(simuLang.contains("nl"))||(simuLang.contains("pl")))
		  notEN=true ;
		  while(1) {
			qDebug() << "searching" << simuName << simuLang << "simulator";
			foreach (QString name, registered_simulators.keys()) {
				if (notEN) {
					if (name.contains(simuName + '-' + simuLang)) {return registered_simulators[simuName=name];}
					} else if (name.contains(simuName)) {return registered_simulators[simuName=name];}
			}
			int pos = simuName.lastIndexOf('-');
			if (pos <= 0)
			  break;
			simuName = simuName.mid(0, pos);
		   }
		  return NULL;
		}

Sacre100, 2015-09-28:
Now as I understand better the structure of the firmware name, may I suggest the following code for getSimulatorFactory in simulatorinterface.cpp

    SimulatorFactory *getSimulatorFactory(const QString &name)
    {
      QString simuName = name;
      QString simuLang = "-en";                                                       // langue par défaut
      if ((simuName.count('-') > 1)) {                                                // si on a plusieurs traits d'union
        simuLang = simuName.mid(simuName.lastIndexOf('-'));                           //   on récupère -<langue>
        simuName = simuName.mid(0, simuName.indexOf('-', simuName.indexOf('-') + 1)); //   on récupère opentx-<firmware>
      }
      if (simuLang != "-en") {                                                        // si la langue est différente de l'anglais
        simuLang = simuName + simuLang;                                               //   on construit opentx-<firmware>-<langue>
        if (registered_simulators.find(simuLang) != registered_simulators.end())      //   si on trouve le simulateur avec la langue
          return registered_simulators[simuLang];                                     //     on renvoie le simulateur trouvé
      }
      if (registered_simulators.find(simuName) != registered_simulators.end())        // si on trouve le simulateur sans la langue
        return registered_simulators[simuName];                                       //   on renvoie le simulateur trouvé
      return NULL;
    }

.

fred99, 2015-10-01:
I don't know if that can help but we have worked on point 1 and point 2 and shared infos on this page :
http://frskytaranis.forumactif.org/t2536p135-compilation-de-companion-2-0-et-2-1-et-du-simulateur-en-fr#31803

fred99, 2015-10-11:
@Kilrah commented "the generation of all languages should ideally be managed by the cmake build environment."
I hope you will like my solution ;) Regards.

The key is the optionnal [binary_dir]

    add_subdirectory(source_dir [binary_dir]
                 [EXCLUDE_FROM_ALL])

File companion\src\CMakeLists.txt
Loop to create simulator-xx build files.
Code  after the line add_subdirectory(simulation):
    
    add_subdirectory(simulation)
    SET (translation FR IT DE EN)
    foreach(TRANSLATIONS  ${translation})
      set (SUFFIX "")
      string(TOLOWER ${TRANSLATIONS} TRANSLATION_SUFFIX)
      set(SUFFIX ${SUFFIX}-${TRANSLATION_SUFFIX})
     add_subdirectory(firmwares/opentx/simulator/ ../build/simulator${SUFFIX})
    endforeach(TRANSLATIONS)  

File companion\src\firmwares\opentx\simulator\CMakeLists.txt 
EN don't use -en in name.
Code at the beginning of the file:

    if (TRANSLATIONS STREQUAL "EN")
        set (TRANSLATIONS "")
        set (SUFFIX "")
        MESSAGE( "Generating -en")
        else ()
        MESSAGE( "Generating " ${SUFFIX})
    endif ()
    set(COMMON_DEFINITIONS ${COMMON_DEFINITIONS} TRANSLATIONS_${TRANSLATIONS})

Tests:
cmake -G "Visual Studio 10" ..\src is OK :
output:
      .... 
      Generating -fr
      Generating -it
      Generating -de
      Generating -en
      -- Configuring done
      -- Generating done
      -- Build files have been written to: C:/Users/SSD/Desktop/machine virtuelle/opentx-2.1.3/companion/build

ALL_BUILD is OK:
output:
    ========== Build: 58 succeeded, 0 failed, 4 up-to-date, 0 skipped ========== 

Companion version @Sacre100 is OK.

Silmulator is OK.

projectkk2glider, 2015-10-11:
@fred99 we would like to see a pull request with your modifications.

fred99, 2015-10-11:
https://github.com/opentx/opentx/pull/2963

LapinFou, 2015-10-23:
I hope this will be added soon! It would be so nice to have multi-language simulator under Companion. :smile:

fred99, 2015-10-26:
The pull request #2963 has been closed; the solution introduced doesn't answer to the ambitious goal to have a multilanguage and multi-options simulator and can't be a candidat for a merge.

LapinFou, 2015-10-26:
Does it mean, it will never be implemented?? It would be a real shame. :cry:

bsongis, 2015-10-28:
@fred99 don't you want to take point 2 I described in a post before (23 Sep)?

fred99, 2015-10-28:
You let me work with Sacre100 in the wrong direction without saying anything that our developments did not match your vision, until the end.
I am aware you are very busy, especially with Horus, so may be that was not intentional but just a missing.
The choice of options on the simulation present no interest for me, the selected set is suitable for me (and I have not seen a complaint on the the french forum), the language is the top priority.
I have a great respect for you and your project that has revolutionized the world of radio and I'm really sorry to have lived such a negative experience with the devs. 
I will continue to make some donations, this is the best I can do.

projectkk2glider, 2015-10-28:
@fred99 I am sorry you feel this way. There has been some sort of miss-communication here ether due to lack of time or something else. The fact is that your work now looks (at least partly) in vain. 

What can I say, stuff like this can happen and will happen in future. We don't have a dedicated team manager here. It is all voluntary work, done by people with real jobs/families/etc in their free time. I hope you will change your mind.

bsongis, 2015-10-28:
The fact is ... I didn't see in which direction you were going before the pull request! By the way the code you wrote is perhaps good (I mean the function to find the best simulator), I still have to check.

Sacre100, 2015-10-28:
If the logic is to compile and download the simulator on demand with the same options as configured for the firmware (including language) in the same way done for the firmware, the code is not appropriate.

It's not an issue for me if the work is lost, it's a good experience with a lot of good point (setup of the build environment, friendly collaboration with Fred, refresh of my C/C++ practice, etc...)

closed 2020-07-07 by stale[bot]

dev-fred, 2020-07-07:
Adieu!
 je vais continuer à compiler les DLL's des simus en français