opentx/opentx #2896: [OpenTX2.1.3][Bug] Model comparison issue: curves

LapinFou, 2015-09-18:
Hi folks,

The new comparison engine is much better, but there is still one issue with the curves.
See the snapshot below:
[image]

projectkk2glider, 2015-09-18:
Please upload an eepe and tell us which models to compare to see this result.

LapinFou, 2015-09-18:
At first, I wasn't able to reproduce the problem, but I finally discovered the right sequence to reproduce the problem.
http://1drv.ms/1KlNjUS
1- Open the comparison windows.
2- drag & drop the T-Rex500 model on the left side.
3- drag & drop the Stratus model on the right side.
4- drag & drop the same T-Rex500 model on the right side this time.
Curves are not refreshed.

fraternl, 2015-09-18:
Does it only happen if you are comparing the same models? It should work then as well, but that might also be the reason it now doesn't work.

LapinFou, 2015-09-18:
To me, it look likes if you drag & drop another model without closing the comparison windows, then the curves plotting is not refreshed.

Ex:
1- Open the comparison windows.
2- drag & drop the T-Rex500 model on the left side.
3- drag & drop the same T-Rex500 model on the right side.
3- drag & drop the Stratus model on the right side.
Curves are not refreshed.
[image]

projectkk2glider, 2015-09-19:
Can not reproduce in current `master` and official 2.1.3. Perhaps it is only a Windows problem? We use the same file names and generate new contents in them. Maybe the display is generated from the cached image somewhere? Again can not reproduce on Linux.

closed 2015-09-19 by bsongis

LapinFou, 2015-09-19:
When the 2.1.4 will be out, I will try again.
Thanks for your support.
:)

projectkk2glider, 2015-09-19:
Or you could try the nightly build tomorrow http://downloads-21.open-tx.org/nightly/companion/

LapinFou, 2015-09-19:
I will !
:)

LapinFou, 2015-09-21:
I confirm the issue has been resolved in the build 2.1.4N465.
Thanks!!

PR #2901 (2015-09-19): Fixes #2896: draw curves without using temporary disk files
Closes #2896.

Perhaps this change will fix the original issue. The model curves are no longer created on disk, but embedded directly into the document.
commit 365d2b6c (2015-09-19): Fixes #2896: draw curves without using temporary disk files
commit d63ac41b (2015-09-19): Merge pull request #2901 from opentx/projectkk2glider/issue_2896_curves_without_files

Fixes #2896: draw curves without using temporary disk files
commit 69a6a185 (2015-09-19): Fixes #2896: draw curves without using temporary disk files (ported from 2.1)