opentx/opentx #2898: Trim switch test affects actual trim

zieren, 2015-09-19:
[X9D+ 2.1.3] In the switch test screen (Menu long->5x Page), operating the trim switches doesn't just test them, it also affects the actual trim of the model. Is that intentional? IMHO it's unintuitive and may lead to users messing up their trim settings, possibly w/o even realizing it.

kilrah, 2015-09-19:
This page is for troubleshooting faulty hardware, you'd obviously not do that with a "real" model selected.

zieren, 2015-09-19:
I was actually troubleshooting my flaky trim switch. Why is it obvious that I need to select a dummy model for that? IMHO the expected behavior on a test page is to not affect "production" data.

projectkk2glider, 2015-09-19:
Maybe there is a simple solution to actually disable trim switch action when the test screen is active? It would be more user friendly indeed.

kilrah, 2015-09-19:
Good luck doing that with all the different possible actions of the trim switch :)

The basic behavior of any R/C radio is that anything you touch always responds and gets stored at any time.

projectkk2glider, 2015-09-19:
>Good luck doing that with all the different possible actions of the trim switch :)

I kind of suspected that :grin: :  Too many possibilities to mess something up. Closing...

closed 2015-09-19 by projectkk2glider