opentx/opentx #2908: Compilaton warning "FrSky module alarms removed!" necessary?

projectkk2glider, 2015-09-20:
There was a `#warning` introduced in commit https://github.com/opentx/opentx/commit/88d9691d16fa96691a74595ed8e17d021c868fd5#diff-146f1098a524de4323eff6d3d2817e61R416
which produces anoying compiler warnings:
```
./../radio/src/telemetry/frsky_d.cpp:416:2: warning: #warning "FrSky module alarms removed!" [-Wcpp]
```

Is there still any reason for it, or could that part of the code be cleaned up now?

bsongis, 2015-09-20:
Yes, nobody complained yet for this part missing, I think we could remove the warning!

bsongis, 2015-09-20:
And then all warnings should become errors, this would be really nice ...

projectkk2glider, 2015-09-20:
Let me do this.

bsongis, 2015-09-20:
Thanks! Also perhaps there will be some eeprom that can be freed in 'next'

projectkk2glider, 2015-09-20:
I am also looking at splitting frsky_d.cpp code into ARM and nonARM files. The current file is hard to read with so many ifdefs. I will make a PR and you will decide if you like it.

bsongis, 2015-09-20:
Oh yes, thanks!

PR #2910 (2015-09-20): Projectkk2glider/issue 2908 frsky d refactoring
Closes #2908.

Please be careful. I only tested compilation of the new code. No tests have been done on the radio! The `-Werror` is enabled only on platforms where it passes (I fixed one problem for Taranis). Other boards fail this and do not have this option enabled (yet).
closed 2015-09-20 by bsongis