opentx/opentx #2913: Trainer mode not work 2.1.4N465 X9E

janekx, 2015-09-21:
1. X9E with 2.1.3 stable slave / jack
2. X9E with 2.1.4N465 master / jack

using 3.5mm mono jack

1. new model slave / jack 1-8ch 22.5 300u -
2. new model master/ jack at main menu can not select calibration. the numbers at 0.0 0.0 0.0 0.0 right aside Cal. very quickly and hard to noticible shows signals from slave. To see -100 to 100 look like shows for 1/200sec only so disapear very quicky. 

When turn roles from radio 1 to 2 and vice versa it is same.

bsongis, 2015-09-26:
Tested this morning on my X9E prototype

- I have the trainer / trainee icon always present in the notification bar (hardware bug fixed on production radios)

- I have the trainer / trainee correctly detected (haptic bzzz)

- **As described above I have 0.0 in all numbers from the Trainer screen**

bsongis, 2015-09-26:
Does anyone has Trainer function correctly working on a production X9E?

janekx, 2015-09-28:
Here is my test, 2 production line X9E with 2.1.3, one first batch X9E with 2.1.4N465 and X9D+ with 2.0.17 
The firstbatch X9E not tell the connection of the trainer port, but production line X9E tells conection and disconnection. On X9E there is no icon of the connected trainer port. 

http://www.youtube.com/watch?v=m_W7bI_TSKQ

arnaudmz, 2015-10-02:
Issue confirmed here, the tests I ran:
  * master: X9D, slave: Spektrum D7 => **OK**
    * Icon appears on X9D **OK**
    * sounds do notify jack connection and disconnection **OK**
    * slave sticks move the servos **OK**
  * master: X9D, slave: X9E => **OK**
    * Icon appears on X9D **OK**
    * sounds do notify jack connection and disconnection **OK**
    * slave sticks move the servos **OK**
  * master: X9E, slave: X9D => **KO**
    * Icon won't appears on X9E **KO**
    * sounds do notify jack connection and disconnection **OK**
    * slave sticks won't move the servos **KO**
  * master: X9E, slave: Spektrum D7 => **KO**
    * Icon won't appears on X9E **KO**
    * sounds do notify jack connection and disconnection **OK**
    * slave sticks won't move the servos **KO**

Both X9D and X9E are running 2.1.3 OpenTX, My X9E might come from the second batch (recieved it just one week ago)

hrenz, 2015-10-03:
Hy,
same Trainer-Problem  

X9E first batch, openTx V2.1.4 N453
X9DPlus, one year old,  openTx V2.1.2 N417
Simple old 4K TX as slave/jacket,  PPM-Signal at Oszi tested, is ok, 4,5V 

Test with Mono and Stereo Kabel

X9E as slave/jacket, X9D as master/jacket,  is ok, all works fine, symbol only at X9D, not at X9E
X9E as master/jacket, X9D as slave/jacket,  not ok, nothing input at X9E no symbol at X9E
X9E as master/jacket, Simple old 4K TX as slave/jacket, not ok, nothing input atX9E no symbol at X9E

Huston we have a problem

Eagle69, 2015-10-04:
Hi

before I did the update of the software from FRSKY original to Open TX the master/slave function did worked, at least that's how I remeber it.
Is this easy to test again ?

bsongis, 2015-10-04:
@Eagle69 That's interesting, are you sure?

Eagle69, 2015-10-05:
@bsongis 
Pretty sure that I could read it on the TX before I changed the Software.
I think I saved FrSky original Sw so I should be able to reverse and try it.

bsongis, 2015-10-05:
Yes, thanks!

Eagle69, 2015-10-05:
Hi

I couldn't restore it but since it works the other way around it can't be Hw error.
Even if I connect a Spektrum Tx to X9E the Spektrum Tx lit up.
Or ??

bsongis, 2015-10-05:
Here we need @janekx I am almost sure he performed tests with the X9E stock firmware, am I right?

janekx, 2015-10-05:
Tested with opentx 2.1.3 and nightly,  never test with fw the FrSky load there.

bsongis, 2015-10-05:
Ok, I need the FrSky original firmware, I will ask Adela.

janekx, 2015-10-05:
They have holiday now.  Will be from 8.10 back in work.  I am possible to test it too...

janekx, 2015-10-05:
Officially they have only lattest available i think we need that older one http://www.frsky-rc.com/download/view.php?sort=Firmware&down=213&file=Firmware-%20Taranis%20X9E

bsongis, 2015-10-05:
@janekx Would you check it it's better with this firmware?

janekx, 2015-10-05:
Ok will try it tomorrow and will report,  but this FW was relased after my batch come to me. So can be fork of the newer opentx that "contain" bug.

bsongis, 2015-10-05:
Not sure, especially if it comes from a different compiler

janekx, 2015-10-05:
I see it is alternative too

Murphy25, 2015-10-05:
Bonjour,

Je viens de tester sur ma X9E non EU avec un câble jack mono.
- Moniteur : X9E non EU firmware 2.1.3
- Elève : Radio Protonik PTR-6A V2

J'ai aussi le problème alors que la double commande fonctionne bien avec ma X9D en 2.0.17.

Merci d'apporter une solution car je compte bien équiper ma X9E en double aussi avec un récepteur connecté en PPM comme sur ma X9D.

GervaisPinot, 2015-10-06:
Bonjour,

J'ai recu ma X9E, j'ai transféré tous mes modèles de la X9D et j'ai un problème avec l'écolage.

L'écolage  ne fonctionne plus. Quand je branche le Jack sur la prise arrière de la radio (la bonne, pas la prise caque....) je ne vois rien dans les signaux d'entrée dans l'écran de calibration de l'émeteur élève. Y a t'il una endroit ou on doit déclarer le mode élève autre part que dans le menu modèle ?
J'hésite entre un problème de soft et un problème de hard.

Est ce que qu'elqu'un a fait fonctionner le mode élève avec une X9E ?

Merci pour vos réponses.

bsongis, 2015-10-06:
Justement c'est l'objet de l'issue, on ne voit personne pour qui ça
fonctionne

janekx, 2015-10-06:
So I tested on production X9E the FW from FrSky DL page. 
Same issue.
It says trainer port connected, no icon, no signal on Trainer setup page. 
Same as before there is lightly see the numbers but only like shadows.

Firmware says 2.1.4 Eeprom 217 
2015-09-24

Murphy25, 2015-10-06:
Bonjour,

En tout cas ce n'est pas un problème de hard puisque dans le sens X9E élève > X9D moniteur ça fonctionne. Donc la prise est bonne.

janekx, 2015-10-06:
Please use English...  Thanks

Murphy25, 2015-10-06:
Sorry, Google my friend :)

janekx, 2015-10-06:
OK tak se tu muzeme zacit bavit cesky a bude ti to stejne k nicemu, protoze to google spatne prelozi. :)

bsongis, 2015-10-06:
He says something wrong. For the moment we can't say anything. PPM in and PPM out are on different PINS on the CPU.

arnaudmz, 2015-10-07:
Bertrand, is there any way we can be of any help investigating this?
I can backup everything and try any kind of debug/experimental firmware if you like.

Let us know.

bsongis, 2015-10-07:
My problem is simple. FrSky told me that they don't have this problem. I need them to double check, if they are right, it means they fixed an hardware problem. If wrong there is a bug in the software. But when you plug the trainer cable the fact is there are a lot of interrupts, it seems the PPM input is working, but only during ~50ms

janekx, 2015-10-07:
I got scope back,  i can do it but do not know where to check have no diagrams to look at.  
50ms looks like that ghost numbers rhat can be lightly seen

bsongis, 2015-10-07:
You will probe on either side of the cable!

janekx, 2015-10-07:
So i should check if the trainee send  the PPM correctly right?

bsongis, 2015-10-07:
In fact you will check if once the wire is connected your PPM stream remains OK. If the X9E grounds the PPM signal, then it is a problem ...

janekx, 2015-10-07:
I see the task...

wollenagold, 2015-10-08:
Hi guys
When I got my X9E 6 days ago I made a security of the original bin. whit which the TX was delivered to me.
Perhaps you need such an original bin.  tell me to what e-mail I can send to help you to fix the bug.
Because I need the opinion teachers-TX while I make the introduction of beginners in our club.
Wolfgang

bsongis, 2015-10-08:
@janekx already told here that the original firmware has the same problem

janekx, 2015-10-09:
As I can measure the pads on the PCB of DSC port there is already good signal on trainer from trainee.

kilrah, 2015-10-09:
Some on RCGroups however said the original FW was OK...

janekx, 2015-10-09:
will try to contact FrSky for that FW, older then we tested ....

GervaisPinot, 2015-10-09:
Hello,

I have checked again the trainer mode  with the initial firmawe delivered
with the TARANIS X9E
opentx-taranis-x9e-2.1.0
EEPR217

The trainer mode works fine., The same way it worked with my X9D

This confirm the firmware issue with the 2.1.3

Hope this helps the problem solving.

Let me know when you wish extra information.

Gervais

arnaudmz, 2015-10-09:
Sounds like good news to me.

hrenz, 2015-10-09:
Hy, 
could I get please the original Frsky file
for the TARANIS X9E opentx-taranis-x9e-2.1.0 EEPR217
for Testing too master slave

janekx, 2015-10-09:
I think there  is possible to save the fw to file and send it to us for test.

GervaisPinot, 2015-10-09:
Here is the initial firmware

gervais

bsongis, 2015-10-09:
@GervaisPinot Yes this is excellent news, would you send this firmware to my email address? bsongis (at) gmail (dot) com

janekx, 2015-10-09:
Me too please to info (@) rcstudio (.) cz

janekx, 2015-10-10:
OK I got from Adela their 2.1.0 FW and I tested it now, it works well, Adela told me that their tests work OK on 2.1.0 and on 2.1.3 have issue. 
So we have confirmation that need to be something corrected in OpenTX and not HW issue.

bsongis, 2015-10-10:
Ok. I will try to find which revision introduced this problem. It's indeed good news.

projectkk2glider, 2015-10-10:
The most suspect is commit f6bff9683dd3ad058c0a5aa16060f08b963cfad8. I bet the new PPM validity check is discarding input signal. 
```c
inline void captureTrainerPulses(uint16_t capture)
{
  static uint16_t lastCapt=0;
  static uint8_t channelNumber=0;

  uint16_t val = (uint16_t)(capture - lastCapt) / 2;
  lastCapt = capture;

  // We process ppmInput right here to make servo movement as smooth as possible
  //    while under trainee control
  //
  // G: Prioritize reset pulse. (Needed when less than 16 incoming pulses)
  //
  if (val>4000 && val<19000) {
    channelNumber = 1; // triggered
  }
  else {
    if ((channelNumber > 0) && (channelNumber <= NUM_TRAINER)) {
      if (val>800 && val<2200) {
        ppmInputValidityTimer = PPM_IN_VALID_TIMEOUT;
        ppmInput[channelNumber++ - 1] =
          //+-500 != 512, but close enough.
          (int16_t)(val - 1500)*(g_eeGeneral.PPM_Multiplier+10)/10; 
      }
      else {
        channelNumber = 0; // not triggered
      }
    }
  }
}
```

projectkk2glider, 2015-10-10:
Using `git bisect` I determined that commit 60f2535212b04d06202e679f11299c16298d03a3 broke this (remember only when **haptic** option is selected).
```
opentx $ git bisect log
# bad: [bb1d368c039ecf5208c0e0f119bcc12c88cef7c3] Merge pull request #2933 from hrenz/patch-11
# good: [6fe12ba6251ac05ed8f38548b487f91645ac0c75] Cosmetics
git bisect start 'bb1d368c039ecf5208c0e0f119bcc12c88cef7c3' '6fe12ba6251ac05ed8f38548b487f91645ac0c75'
# good: [1d6adcc25a34c8f02b6406b215f58e366ff1bd1a] Fixes #2870
git bisect good 1d6adcc25a34c8f02b6406b215f58e366ff1bd1a
# good: [3f6e1d0b3a36857371b67e31e57a4c8c5a6cba8d] Lua error display function reused
git bisect good 3f6e1d0b3a36857371b67e31e57a4c8c5a6cba8d
# bad: [7ec4f508eec571e8331d25c50fa2596ce099384d] Merge pull request #2930 from opentx/bsongis/issue1914_by_dsbeach
git bisect bad 7ec4f508eec571e8331d25c50fa2596ce099384d
# bad: [60f2535212b04d06202e679f11299c16298d03a3] Fixes #1914
git bisect bad 60f2535212b04d06202e679f11299c16298d03a3
# first bad commit: [60f2535212b04d06202e679f11299c16298d03a3] Fixes #1914
```

bsongis, 2015-10-10:
Yes it is the most suspect, but there is no difference in the validity check itself!

bsongis, 2015-10-11:
No luck, I have another problem on my X9E prototype, the PPM signal is grounded if the plug is in its place. I have to pull a little bit the cable to make it work (with the firmware I received by email (date: 15 july of 2015)

janekx, 2015-10-11:
It look more like the mechanical issue the jack is not right in place. I found that there the jack female on PCB no go thrue the case and can make this issue....

bsongis, 2015-10-11:
I have found

bsongis, 2015-10-11:
Don't be too hard with me when you see the commit please ...

bsongis, 2015-10-11:
I just forced a new nightly build. Please check if it's ok, but I am confident, it works here (except the mechanical issue)

janekx, 2015-10-11:
N485 and still no data, appears -1.6 0.2 0.2 1.0 but not change, trainer port cennected and disconnected it tells.

janekx, 2015-10-11:
N486 on production line work for me well. Thanks

hrenz, 2015-10-11:
Hy,
Now I have the N486 at X9E too  

X9E als Master/Jacket  and a X9DPlus as Slave/Jacket

first problem is/was  the jacket !! Yes there is a contact-problem!!

second  
master/ jacket as input now is ok, can see the values from trainer comming
haptik comes at putting the jacket in

but 
no symbol at X9E main screen,  shouldn´t there be a trainer symbol too, same symbol as at X9D
no speach if Trainer signal is going On or Off, sholdn´t it be a speach?

[46 more replies]

commit 30f5b1b2 (2015-10-11): Fixes #2913
closed 2015-10-16 by bsongis