opentx/opentx #2914: [OpenTX 2.1.x][Wishes]Added W*min unit for telemetry sensor

LapinFou, 2015-09-21:
Hi folks,

I'm using the great UniSens-E sensor.
FYI, here are the send telemetry data of this sensor (FW v1.11):
Id 0110: vertical speed
Id 0100: altimeter
Id 0500: rpm
Id 0210: VFas
Id 0600: Fuel level (LiPo capacity must be programmed into the sensor with its USB dongle)
Id 0410: Send as temperature, but is in fact a energy value

So, since the version 1.11, it send the energy on the Temp2 channel (Id: 0410).
The unit is W*min (it is "times", not "per" minute). So it must be multiply by 60 to get the Joule unit (W*sec).

Would it be possible to add the W*min unit?

Thanks!!

LapinFou, 2015-09-21:
I realized, it is a bit more complex as expected, because you should also add the corresponding system voice announcement.

pindeslandes, 2015-09-22:
Hi LapinFou,

You might be interested in this one: #2818.
Though I am afraid it didn’t raise as much enthusiasm as I was expecting among the devs, maybe you could find some better suggestion to make from it.

LapinFou, 2015-09-22:
From my understanding, the only solution is to request to the devs to add the missing units. Anyway, I don't think there are so much missing units...

kilrah, 2015-09-22:
I'd be OK to add the unit, but at 2 conditions:
- The Unisens developer talks to FrSky and requests a specific ID for that purpose instead of misusing the TEMP field. One of the main goals with the new telemetry handling in 2.1 was to get rid of that mess.
- They change to Joules (W*s) as it just makes no sense to use a nonstandard unit.

LapinFou, 2015-09-23:
I will contact them.
Concerning, the unit, maybe they are using W*min to avoid too large number → W*min * 60 = J (W*sec)

projectkk2glider, 2015-09-23:
That was true when he had to stuff the Wmin value into temperature. With the new ID he will be able to use uint32 type (maximum 4294967296)

LapinFou, 2015-09-27:
I got an answer from SM-Modellbau. Here it is:
> a specific ID is no problem. But the used unit must be "W\*min", as
> this value is specifically for use in F5 competitions and the rules
> refer to "W\*min".
So, concerning the 2 conditions requested by Kilrah, the 1st will probably done through an UniSens-E firmware update, but the 2nd one will not be done.

fraternl, 2016-11-14:
@projectkk2glider Hi Damjan.... I think this one also needs the label "EEprom change"

closed 2020-07-08 by stale[bot]